moveit/moveit_task_constructor #235: Add time benchmark for collision detection

captain-yoshi, 2021-02-18:
I think it would be great if MTC could add some information about how much time is spent in the collision algorithm for every solution of every stages that uses it. In motion planning, collision detection is often the bottle neck. 

Things to consider:
- Add a metric for the number of discrete states the collision detection triggered (To calculate the mean of time per state)

This would be usefull for many things:
- Comparing multiple collision detections
- Compare mesh quality
- Compare mesh quantity

I don't think this would require much change from MTC. But for MoveIt it maybe complex. If you say that it would be feasible and beneficial, I could try and implement it.

Thx again for this amazing framework!

v4hn, 2021-02-18:
Hello there,

collision detection depends solely on the algorithm of the respective stage (which is user-provided from the point of view of the framework).
So this information can't be collected/provided by the framework in general, though we could in theory provide Stage methods to register that information from within the stage.

Of course you could, for example and assuming you rely on MoveIt's FCL-based collision testing, add a global timer [to the collision callback](https://github.com/ros-planning/moveit/blob/master/moveit_core/collision_detection_fcl/src/collision_common.cpp#L55) and read it after each stage is processed.

captain-yoshi, 2021-02-18:
Thanks for the clarification. 

For CHOMP, STOMP and TrajOpt I'm not sure if collision is decoupled from the planning because I think collision is calculated as an heuristic. So yeah, not as straightforward as I first thought and, like you said, each framework would have to add methods to retrieve timing information. And MTC would have to support each of those, so that would not scale well.

> Of course you could, for example and assuming you rely on MoveIt's FCL-based collision testing, add a global timer

I will do this to be able to do benchmarks. If I was using non random planners, I could easily compare the timing given by MTC between high quality mesh and rough meshes.

captain-yoshi, 2021-02-21:
@v4hn Do you know of any deteministic planners that are suported in MoveIt because this would be helpfull for benchmarking collision (to compare apples with apples)? In the docs I only see `SBPL` but it has not been yet reintegrated in moveit ([MoveIt issue 101](https://github.com/ros-planning/moveit/issues/101)).

v4hn, 2021-02-22:
Well, you can probably go forward and add deterministic seeds to all random number generators in OMPL and the moveit_ompl_interface, but that would restrict the capabilities a lot. The preferred approach should be to benchmark on a statistical basis with many random reruns.

The best deterministic planner in MoveIt is probably [pilz_industrial_motion_planner](https://github.com/ros-planning/moveit/tree/master/moveit_planners/pilz_industrial_motion_planner), but it's a trajectory generator and thus an incomplete planner. So it does not solve many transit motions common in task planning.

v4hn, 2021-03-22:
We agree there is no clear way to add something like that to MTC, so I will close it for now.
Feel free to post again with further ideas.

closed 2021-03-22 by v4hn