moveit/moveit_task_constructor #236: Q: Planning fails without any debug information

dnandha, 2021-02-19:
This is a general question on how to debug the planning stages in mtc:

I'm using a robot configuration different from Panda and even a simple planning task such as moving in Z by -0.2 fails (the robot configuration works fine in the moveit planning interface). No exception is thrown and no debug information is printed, only the return value of `.plan()` is False. Is there any way how I could debug the situation?

rhaschke, 2021-02-19:
The suggested method to debug is to inspect the plan in rviz. As long as you keep the MTC ROS node alive, you can inspect all stages visually in rviz. Did you try this?

v4hn, 2021-03-22:
Closed by timeout. Feel free to write again if you have further questions.

closed 2021-03-22 by v4hn