moveit/moveit_task_constructor #242: Task visualization segfault when rendering failed connect stage

captain-yoshi, 2021-03-10:
When I try to visualize a connect stage that has **failed**, I get a segfault on this [line](https://github.com/ros-planning/moveit_task_constructor/blob/master/visualization/visualization_tools/src/task_solution_visualization.cpp#L468). The variables are `index = 1` and `waypoint_count = 0`. It works for other stages. This seems to suggest that there are no end scenes thus blowing up.
```cpp
(gdb) bt
#0  0x00007fffa26d653a in std::__shared_ptr<planning_scene::PlanningScene const, (__gnu_cxx::_Lock_policy)2>::operator=(std::__shared_ptr<planning_scene::PlanningScene const, (__gnu_cxx::_Lock_policy)2> const&) (this=0x7fffffffb0e0) at /usr/include/c++/9/bits/shared_ptr_base.h:1080
#1  0x00007fffa26d658d in std::shared_ptr<planning_scene::PlanningScene const>::operator=(std::shared_ptr<planning_scene::PlanningScene const> const&) (this=0x7fffffffb0e0)
    at /usr/include/c++/9/bits/shared_ptr.h:103
#2  0x00007fffa23532de in moveit_rviz_plugin::TaskSolutionVisualization::renderWayPoint(unsigned long, int) (this=0x55555566c670, index=1, previous_index=0)
    at /home/captain-yoshi/ws/ros/catkin_ws/src/moveit_task_constructor/visualization/visualization_tools/src/task_solution_visualization.cpp:475
#3  0x00007fffa235303f in moveit_rviz_plugin::TaskSolutionVisualization::update(float, float) (this=0x55555566c670, wall_dt=0.0247225594, ros_dt=0.0247234311)
    at /home/captain-yoshi/ws/ros/catkin_ws/src/moveit_task_constructor/visualization/visualization_tools/src/task_solution_visualization.cpp:437
#4  0x00007fffa26b9fb7 in moveit_rviz_plugin::TaskDisplay::update(float, float) (this=0x555556a94ad0, wall_dt=0.0247225594, ros_dt=0.0247234311)
    at /home/captain-yoshi/ws/ros/catkin_ws/src/moveit_task_constructor/visualization/motion_planning_tasks/src/task_display.cpp:182
#5  0x00007ffff7e76d1b in rviz::DisplayGroup::update(float, float) () at /opt/ros/noetic/lib/librviz.so
#6  0x00007ffff7f3f439 in rviz::VisualizationManager::onUpdate() () at /opt/ros/noetic/lib/librviz.so
#7  0x00007ffff73e2300 in QMetaObject::activate(QObject*, int, int, void**) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#8  0x00007ffff73ef51e in QTimer::timeout(QTimer::QPrivateSignal) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007ffff73e2cf5 in QObject::event(QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x00007ffff77e3a66 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#11 0x00007ffff77ed0f0 in QApplication::notify(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#12 0x00007ffff73b693a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#13 0x00007ffff740d8b0 in QTimerInfoList::activateTimers() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007ffff740e19c in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#15 0x00007ffff463517d in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#16 0x00007ffff4635400 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#17 0x00007ffff46354a3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#18 0x00007ffff740e565 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#19 0x00007ffff73b54db in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#20 0x00007ffff73bd246 in QCoreApplication::exec() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#21 0x000055555555860d in main ()
```

Happy MoveIt day!

v4hn, 2021-03-10:
It sounds reasonable to me to assume that failed connect solutions should have at least one waypoint (the expected end state that could not be connected to, but that assumption seems to fail here.
Is this a matter of which planner you use in Connect or is it independent of the planner?

captain-yoshi, 2021-03-10:
I was 11 commits behind MTC and 8 behind MoveIt. After pulling these, there are no more segfaults. Sorry for opening the issue...

captain-yoshi, 2021-03-10:
@v4hn Finally the bug is still there, I was checking an IK pose and thought that is was working...

> Is this a matter of which planner you use in Connect or is it independent of the planner?

Failing for OMPL and CHOMP.

rhaschke, 2021-03-10:
It's weird that this issue turns up now. For sure, we have seen failed connect stages in the past, which didn't crash rviz...
Could you provide a _minimal_ example to reproduce this issue? Thanks.

captain-yoshi, 2021-03-10:
Tested with the default MTC demo. I lowered the timeout for the Connect stage to 0.01 seconds to force failures. Latest master branch of MTC and MoveIt.

```sh
roslaunch moveit_task_constructor_demo demo.launch
roslaunch moveit_task_constructor_demo pickplace.launch
```

captain-yoshi, 2021-03-10:
@rhaschke @v4hn Sorry for wasting your time... I was on my own custom branch of MTC and not the upstream master branch. I'm closing this because it works as intended. It only happens in my custom branch. My apology.

closed 2021-03-10 by captain-yoshi