moveit/moveit_task_constructor #244: Move to a relative pose

gecastro, 2021-03-16:
Hi,

Is it possible and do you have an example to use a relative movement as the last element of the pipeline?

For example, in the pick and place demo, I would like to:
- stop just after the approach (movement relative to the grasping pose)
- execute some other custom operation, such as visual servoing

So, how can I tell the pipeline which pose to use to calculate the relative movement, but without executing the last pose?

Thanks

rhaschke, 2021-03-16:
The approach motion in pick is computed backwards from the desired grasp pose, i.e. in reverse direction.
Thus, the desired grasp pose is always reached.
Of course, you can also plan a relative movement in a forward direction, starting from some known pose.

v4hn, 2021-03-18:
Basically you have two options.

- you create/configure a stage to generate a pose relative to your goal (nobody says your GeneratePose stages have to specify the goal of the task)

- you compute a whole pick solution, but only execute the segments up to the relative approach motion and then do something else.
[Task::execute](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/src/task.cpp#L317) is a trivial wrapper around a ROS action client, so you can just edit the messages before sending them for execution.

closed 2021-03-18 by v4hn