moveit/moveit_task_constructor #247: Pruning across container boundaries

v4hn, 2021-03-23:
In #221 we implemented correct pruning of infeasible candidate solutions inside SerialContainers.

However, this does not work across container boundaries yet because Containers currently act as opaque stages in their parent.

Consider this example structure:

[image]

Assuming InterfaceState `a` fails to extend in Stage `Back`, `Fwd2` does not have to compute  InterfaceState `b` and the state can be disabled.
The same is true for the inverted case.

Parallel containers can result in similar situations but have to consider alternative paths through the different children as well.

I believe this is what you meant in https://github.com/ros-planning/moveit_task_constructor/pull/221#issuecomment-804847964 @rhaschke ?

rhaschke, 2021-03-24:
Yes, that's definitely an open issue. I think, the other way around, i.e. a failing InterfaceState inside der container isn't yet propagated to the outside either.

rhaschke, 2021-03-29:
I think we need to rethink the propagation of the `Interface::Status` for pruning once again: Currently, propagation occurs along connected solutions, neglecting the type of stage owning the `Interface`. Thus, currently, we cannot propagate through `Connect` stages, if their solutions are not yet connected. However, here pruning would be particularly beneficial, because we can rule out many opposite InterfaceStates. I think, status propagation should be handled by the (private?) stage classes.

Also, in order to allow propagation from inside containers to the outside, containers need to hook into their children's Interface to get notified on Status changes. This is currently not possible either.

PR #252 (2021-03-31): prune outside states on failure inside container
Followup to #250 and should fix #247 (up to the additional pruning you mentioned in the last post @rhaschke).
closed 2021-03-31 by rhaschke