moveit/moveit_task_constructor #248: Copy & Move Constructors for Stages?

v4hn, 2021-03-25:
Currently [`Stage`'s copy constructor is deleted](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/include/moveit/task_constructor/stage.h#L251) and as a result move construction is not available either.

The main reason for the deleted constructor is that we handle a lot of raw pointers in the structures, especially after adding it to a container/task and copying would be quite fragile.

However, I don't really see a reason why we shouldn't copy or move stages when they are not connected to a task structure yet.
Corresponding constructors should be straight-forward to implement.

rhaschke, 2021-03-25:
I'm confused. Aren't we using move construction to move stages into a container?

v4hn, 2021-03-25:
No, we only ever move `unique_ptr<Stage>`, not the stages themselves.