moveit/moveit_task_constructor #256: Make pruning optional for debugging

v4hn, 2021-04-23:
We recently reworked pruning (#221, #250, #252). It works more reliable now and can reduce computational overhead quite a lot.
However, for debugging, it can be useful to attempt all possible computations.
This can help to identify culprits and identify longest partial solutions in the current design.

I would propose to add a `pruning` property to the [`Task`](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/include/moveit/task_constructor/task.h#L111) container and allow users to call `setPruning(false)` if they want to do so.

rhaschke, 2021-04-24:
Maybe, instead of defining this property in the top-level task container only, you want to do so at the individual container level?
I also suggest automatically inheriting this property from the parent.

v4hn, 2021-04-24:
> Maybe, instead of defining this property in the top-level task container only, you want to do so at the individual container level?
> I also suggest automatically inheriting this property from the parent.

Yes. I don't think we could implement it easily any other way, because the individual stages need to be able to see the flag.
I think there are also unresolved issues with default sources for inheritance though - I seem to recall we don't support them at the moment.

rhaschke, 2024-07-06:
Fixed via #590

closed 2024-07-06 by rhaschke