moveit/moveit_task_constructor #257: feedback from incompatible Connect attempts

v4hn, 2021-04-23:
If incoming pairs in the `Connect` stage [are incompatible](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/src/stages/connect.cpp#L100), these pairs are silently dropped.
But as a result it is impossible to debug in RViz when all incoming pairs are incompatible.
This is a rather usual problem during setup if for example a `MonitoringGenerator` is connected to an incompatible generator stage.

So we should spawn failures to connect as well, but because of the combinatorial explosion the number of states might be too much to handle.
Some approaches are:
- Just try the naive approach
- Spawn only a filtered subset of failures. Caching by cause might add a lot of overhead, but could be the best way to go
- Limit the number of spawned failures

It might also make sense turn [`storeFailures`](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/include/moveit/task_constructor/stage_p.h#L149) into a property to let users decide to explicitly switch it off for `Connect` stages.

rhaschke, 2021-04-24:
> Some approaches are:
> * Just try the naive approach

The naive approach would be to report all incompatible pairs? :-1:

> * Spawn only a filtered subset of failures. Caching by cause might add a lot of overhead, but could be the best way to go

:+1:

> * Limit the number of spawned failures
> 
> It might also make sense turn [`storeFailures`](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/include/moveit/task_constructor/stage_p.h#L149) into a property to let users decide to explicitly switch it off for `Connect` stages.

:+1:

v4hn, 2021-04-24:
> The naive approach would be to report all incompatible pairs? -1

I agree it's not a nice approach, but it's pretty much what I do when I print ROS_INFO lines for each pair to debug such issues.

Caching by cause would be great, but is much more complicated because we need to add hashable causes for all compatibility tests.