moveit/moveit_task_constructor #258: Show partial solution paths in RViz

v4hn, 2021-04-23:
Whereas users can click through individual solutions of a stage, they can't yet see the connected solutions of other stages.
It would be very useful to get the following GUI behavior:

- clicking on a solution in one stage will set the background color of all stages that contain an solution belonging to the same solution path. Bonus points for displaying the failed/pruned stages in other colors.
- clicking on any other highlighted stage as a follow-up will highlight the individual associated solutions there so users can inspect them easily

rhaschke, 2021-04-24:
> * clicking on a solution in one stage will set the background color of all stages that contain an solution belonging to the same solution path. Bonus points for displaying the failed/pruned stages in other colors.

In the usual case of serial containers, _all_ stages will be highlighted if a full solution exists, right?
I think, the only interesting use case is to highlight related failed/pruned stages that caused this particular solution to become _not_ part of a full solution, i.e. _only_ handle the situation you claimed as bonus.

> * clicking on any other highlighted stage as a follow-up will highlight the individual associated solutions there so users can inspect them easily

This is on my wishlist for a long time. I wouldn't restrict this to highlighted stages though. Switching the selected stage should, in general, also activate the solution(s) of this stage that are related to the active solution of the previously selected stage.

v4hn, 2021-04-24:
> In the usual case of serial containers, all stages will be highlighted if a full solution exists, right?

Right, I wouldn't mind though.
We could of course still filter and only show these highlights if there is no top-level solution that includes it.
I think I would like to see both approaches to decide.