moveit/moveit_task_constructor #262: Missing interface to deserialize Properties

v4hn, 2021-05-03:
I've recently spent a bit of work to improve deserialization and introspect [a stages Propertys](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/include/moveit/task_constructor/properties.h).

I'm struggling with a nice interface though.

For deserialization to be useful, the actual type has to be specified somehow by the user.
That makes any interface for `PropertyMap` (as a datastructure to match `moveit_task_constructor_msgs::Property[]`) rather cumbersome. Either the user adds `PropertySerializer<T>()` for each type they expect (what happens if the stage adds a new type later?) or we have to provide a set of registry entries beforehand, possibly in a standalone library. The latter is probably important anyway because all non-ROS-Message types have to be defined by hand (, e.g. `map<string, double>`).

Additionally a `Property` does not know it's name, but is currently only a component of a `PropertyMap` and not meant to be used standalone. I'm unsure whether it makes sense to change this, but users should be able to deserialize individual properties if they looking for specific information (e.g., which part of a suction gripper to enable).

Additionally, `Property/Map` maintains information on default values and inheritance that is not exported in the messages at the moment. So either we
(1) add that information to the messages as far as possible (not possible to export c++ functions for non-trivial mapping
(2) leave the deserialized `Property` without the missing information
(3) provide a `PropertyView` that only makes the information in the messages available, without being a full representation of the internal `Property`.

Feedback is welcome.

rhaschke, 2021-05-30:
Sorry, I just came across this pending question, when looking through my email stack. Let's start with your last question:
Currently, we only serialize/deserialize the actual value of a property (i.e. option (3) and I don't see the need to change this). The user is interested in the current value only, not where it comes from.

Regarding your main questions: The key problem of serialization/deserialization is that both ends know about the type of the serialized data structure. As far as I remember, this type information is serialized as well. Based on this (text) info, both ends can lookup the corresponding serializer/deserializer (which of course should match) in a common registry.

Best, we have a phone chat to clarify further questions.

v4hn, 2021-05-30:
> Currently, we only serialize/deserialize the actual value of a property (i.e. option (3) and I don't see the need to change this). 

I tend to agree, though it would be nice to reuse the same Property class internally and for deserialization on the client/controller side.
I guess we can still share the same class for these purposes with some refactoring.

> The user is interested in the current value only, not where it comes from.

It somewhat boils down to whether or not we want to be able to reconstruct a whole Task from messages (thus being able to store them as well).
If we want that, we need to add at least the initialization information.
Now that I think about it more though, it's currently not possible to do that anyway because [we allow arbitrary functions](https://github.com/ros-planning/moveit_task_constructor/blob/master/core/include/moveit/task_constructor/properties.h#L128-L129) for the initialization.

> Based on this (text) info, both ends can lookup the corresponding serializer/deserializer (which of course should match) in a common registry.

Yes, I believe we will want to establish another separate library for this registry to implement the "common" in your statement.
At the moment, the respective stages add their types to the registry of the running process via `PropertySerializer`.

> Best, we have a phone chat to clarify further questions.

I will (hopefully) be available on the Discord server the whole next week.
Please send a mail if you intend to join (I believe there might be more than that to discuss ;)).