moveit/moveit_task_constructor #276: Set desired/max Cartesian end-effector speed

gautz, 2021-06-23:
[``setMaxCartesianEndEffectorSpeed()`` should soon be available through the MoveGroup interface](https://github.com/ros-planning/moveit/pull/2674). Which would be the proper way to get a similar functionality in e.g. a moveTo stage? Is there some plan on integrating this?

This is following my issue on using MTC for welding #251

rhaschke, 2021-06-24:
Velocity-related stuff is not configured in individual stages, but via the properties of a `PlannerInterface`:
https://github.com/ros-planning/moveit_task_constructor/blob/ee6c50ad31402b00fc4b8a0d8fcd62b1c85c27b7/core/src/solvers/planner_interface.cpp#L47-L48

These are finally used to configure the `MotionPlanRequest`:
https://github.com/ros-planning/moveit_task_constructor/blob/4fa8c10f4456e893e09212b51a06b3d2ec4e9266/core/src/solvers/pipeline_planner.cpp#L150-L151

However, as these settings are only passed to pipeline planners, they will not become effective for Cartesian motions!

v4hn, 2021-06-25:
> However, as these settings are only passed to pipeline planners, they will not become effective for the CartesianPath planner!

So basically, the max Cartesian speed parameter can be added to the `PipelinePlanner` straight away and *all* standard velocity parameters *could* be added to the other solvers as well. Adding the EEF Cartesian velocity to `CartesianPath` is quite a good idea I think.

Also, I would provide a `WrapperBase` container that can adapt all three time parameters for arbitrary stages.