moveit/moveit_task_constructor #278: Compose tasks in rviz GUI

rhaschke, 2021-06-29:
In the very early days of MTC, I strived for a GUI in rviz to compose tasks and make the stages' properties available via nice interactive markers. In response to @JafarAbdi https://github.com/ros-planning/moveit_task_constructor/pull/271#issuecomment-870652402, I will try to summarize the architecture of this approach:

- The rviz MTC panel allows for both [_local_](https://github.com/ros-planning/moveit_task_constructor/blob/8faba159f975cc6da77312293b794be21c8e15f5/visualization/motion_planning_tasks/src/local_task_model.h) and [_remote_](https://github.com/ros-planning/moveit_task_constructor/blob/cf5031e89df984b4ba2bd799864f4a6c5efb671a/visualization/motion_planning_tasks/src/remote_task_model.h) tasks to be shown. 
  - Remote tasks are those running in external applications, communicating their task structure via a [`TaskDescription.msg`](https://github.com/ros-planning/moveit_task_constructor/blob/c471879b08a5a2471ad1deea84160a86f16adef3/msgs/msg/TaskDescription.msg).
  - Local tasks can be composed and modified in the rviz GUI.
- Composition of tasks works via [drag-and-drop](https://github.com/ros-planning/moveit_task_constructor/blob/fc0abe9dd9603ef7fe743641f6d45684e2d8465d/visualization/motion_planning_tasks/src/local_task_model.cpp#L199) from a list of [available (plugin) stages](https://github.com/ros-planning/moveit_task_constructor/blob/84d668b0d5d56c7ba6ce09b919e3763f6c87ebf9/core/motion_planning_stages_plugin_description.xml) via a [StageFactory](https://github.com/ros-planning/moveit_task_constructor/blob/8faba159f975cc6da77312293b794be21c8e15f5/visualization/motion_planning_tasks/src/pluginlib_factory.h).

  https://user-images.githubusercontent.com/5376030/123827977-605f8700-d901-11eb-98c4-d3bcbb5b972b.mp4

- The properties of stages are visualized (in both cases) via an [`rviz::PropertyTreeModel`](https://github.com/ros-planning/moveit_task_constructor/blob/437cc550f2a68895de56f025e2507770bff2faca/visualization/motion_planning_tasks/src/task_list_model.h#L109). 
- Properties are handled by the [`PropertyFactory`](https://github.com/ros-planning/moveit_task_constructor/blob/ee6c50ad31402b00fc4b8a0d8fcd62b1c85c27b7/visualization/motion_planning_tasks/properties/property_factory.h). There are two methods to create a stage's properties: either [via a registered factory method](https://github.com/ros-planning/moveit_task_constructor/blob/ee6c50ad31402b00fc4b8a0d8fcd62b1c85c27b7/visualization/motion_planning_tasks/properties/property_factory.h#L112) for a [specific _stage type_](https://github.com/ros-planning/moveit_task_constructor/blob/ee6c50ad31402b00fc4b8a0d8fcd62b1c85c27b7/visualization/motion_planning_tasks/properties/property_factory.h#L109) or via a default method, operating on the [registered _property types_](https://github.com/ros-planning/moveit_task_constructor/blob/ee6c50ad31402b00fc4b8a0d8fcd62b1c85c27b7/visualization/motion_planning_tasks/properties/property_factory.h) and providing read-only access. Currently, only the second approach is used. The first one was intended to allow for specific property trees, which require the coupling of multiple interrelated properties, e.g. to show a complex interactive marker.

Any help to finish this GUI is highly appreciated.

v4hn, 2021-06-29:
Some issues that are not tackled in either implementation (from what I can see) are
- monitoring stages are not modeled but are essential for all non-trivial tasks
- connections can only succeed between matching types, but during drag and drop the types are not always clear

A closely related project we discussed back then, but that is not implemented until now, is a full serialization of the task.
After all you would not want to run your Task from the GUI most of the time.
@rhaschke Implemented a big step of this later in property serialization to inspect them in RViz, but property-deserialization is not well-supported. The latter is on my close-range agenda and I started rearranging code, but got sidetracked as always..

Personally I prefer to specify Tasks in C++ because it's so much more flexible (e.g., some stages can use C++ methods and it's cumbersome to provide those in a format where specifications can use them), but I do see the merits of a GUI editor.

We agreed before that RViz support for editing properties is quite essential, which is why the unused factory methods are in place.

I would love to see work on the GUI creator continued as well.

rhaschke, 2021-06-30:
- I think full serialization of a task is not meaningful: As you pointed out, there are some features relying on C++ methods or lambdas, which cannot be serialized.
- I agree that property deserialization was an open issue. I remember that we wanted to have a phone call on this. Sorry, I didn't manage yet. Let's try again!
- I think, the rviz GUI should be capable of generating C++ and/or Python code for the preconfigured task. I think that's what @JafarAbdi had in mind with his editor as well?
- Another open issue we discussed before is: Should we allow property editing for remote tasks? If so, how to communicate changes back to the task application? 
- Monitoring stages could be handled by some extra linking mechanism. But that's not yet implemented.
- Drag-and-drop doesn't care about matching connection types and I think it should **not**. If you would forbid non-matching stage sequences, how would you incrementally build a `GEN - CON - GEN` sequence? 
The correctness of the whole pipeline is eventually checked during initialization. On errors, we could highlight corresponding erroneous stages.

v4hn, 2021-06-30:
> I think full serialization of a task is not meaningful

I'm not opposed to a yaml/whatever serialization, though it is indeed quite a lot of work to get this to a useful state and even then it has limits and invokes a lot more interface modelling that would need to be discussed.

> As you pointed out, there are some features relying on C++ methods or lambdas, which cannot be serialized.

Cannot *easily* be serialized. The appropriate solution would be to provide function hooks in shared objects (e.g., via pluginlib).
But although that can work reasonably well it is quite cumbersome especially for testing and development.

> I agree that property deserialization was an open issue. I remember that we wanted to have a phone call on this. Sorry, I didn't manage yet. Let's try again!

I already started working on it (no useful commits though) and plan to continue in the near future. I believe at the moment I know how I want to handle things after our brief exchange in the issue.

> Another open issue we discussed before is: Should we allow property editing for remote tasks?
> If so, how to communicate changes back to the task application?

I would prefer such an API a lot over running Tasks in the RViz process. But then you probably want the whole set: services for modifying the Task structure, for setting properties, for initializing a task, for generating code/a serialization for it, trigger planning, ...
But especially because such a Task can even segfault with custom stages (user code), this approach seems overengineered and fragile to me for insufficient merit.

> Monitoring stages could be handled by some extra linking mechanism. But that's not yet implemented.

Of course. I just wanted to point out that it's not supported at the moment

> Drag-and-drop doesn't care about matching connection types and I think it should **not**. If you would forbid non-matching stage sequences, how would you incrementally build a `GEN - CON - GEN` sequence? 

Yes. Still, interactive feedback would be good to have during the process. If you try to initialize the Task with each modification, you can already get that feedback, but logic on top could also be implemented.

rhaschke, 2021-07-01:
> > Another open issue we discussed before is: Should we allow property editing for remote tasks?

> I would prefer such an API a lot over running Tasks in the RViz process. But then you probably want the whole set: services for modifying the Task structure, for setting properties, for initializing [...]

I agree. Due to the mentioned complexity, I did _not_ planned for such an API. I just planned to display properties of remote tasks in a read-only fashion. One major problem with such an envisioned API is that rviz cannot know about the state of the external application. Thus we don't know whether it is safe to change properties, (re)trigger tasks, etc. 
My intention was to have a way of quickly assembling a task in rviz, primarily to replace and advance the existing MotionPlanning plugin. Optionally, one could generate C++ code from the rviz task to further tweak the code.