moveit/moveit_task_constructor #281: Missing moveit_task_constructor_msgs/SampleGraspPosesAction.h

marcpujol-leitat, 2021-07-12:
I am trying to follow the moveit tutorials to install Moveit Deep Grasps but compiler fails saying this header is missing.

Is this intended and I need to replace them? 

Thank you!

rhaschke, 2021-07-12:
Unfortunately, we didn't yet merge #196, which is the basis for the mentioned tutorial. Also, in https://github.com/ros-planning/moveit_task_constructor/pull/196#discussion_r592675803 it was suggested to use an existing action instead of a new one. This essentially replaced `SampleGraspPosesAction` with [`grasping_msgs::GraspPlanningAction`](https://github.com/mikeferguson/grasping_msgs/blob/ros1/action/GraspPlanning.action). So here is what I suggest:
1. Merge #196 into your MTC repo
2. Update the demo to use `grasping_msgs::GraspPlanningAction`
3. Fix all remaining issues.
4. File a PR against https://github.com/PickNikRobotics/deep_grasp_demo and/or https://ros-planning.github.io/moveit_tutorials to incorporate all required changes into the demo/tutorial.

qyp-robot, 2021-07-12:
Hello, when I was reproducing the GPD algorithm, I followed the Deep Grasp Demo operation, but I encountered a missing header file error when compiling, how can I solve it? Hope you can help me! Thank you very much!
Looking forward to your reply, thank you!
Please be more detailed, thank you

rhaschke, 2021-07-12:
@qyp-robot: Didn't my suggestion point you in the right direction? If not, please provide a detailed error description (error messages, your environment, etc.)

qyp-robot, 2021-07-13:
Hello, do I modify it like this here, just the following two links：
1.https://github.com/ros-planning/moveit_task_constructor/pull/196/commits/a2988995bcb85a099ca6766c709b90eac40d4432
 2.https://github.com/ros-planning/moveit_task_constructor/pull/196/commits/19fb72c81796303336d399905576a7de69602fc0

qyp-robot, 2021-07-13:
Hello, when I make changes according to this, it can be compiled normally;
But after I ran roslaunch moveit_task_constructor_gpd gpd_demo.launch, moveit did not capture, and the following error was reported on the terminal:

[ INFO] [1626148048.631361699]: Generate grasp goal now active
 runtime(computeNormals): 0.3677
camera: 0, #indices: 3998, #normals: 3998 
Calculated 3998 surface normals in 0.3699s (mode: OpenMP).
Reversing direction of normals that do not point to at least one camera ...
 reversed 0 normals
 runtime (reverse normals): 4.5744e-05
Estimating local reference frames ...
Estimated 50 frames in 0.0006s.
Finding hand poses ...
Found 50 hand sets in 0.32s
====> HAND SEARCH TIME: 0.321938
Generated 50 hand sets.
Filtering grasps outside of workspace ...
Number of grasp candidates within workspace and gripper width: 400
Number of grasp candidates with correct approach direction: 400
neighborhoods search time: 0.0382
Created 400 images in 1.1584s
[grasp_cloud_detection-2] process has died [pid 410550, exit code -11, cmd /home/qyp/ws_grasp/devel/lib/moveit_task_constructor_gpd/grasp_cloud_detection __name:=grasp_cloud_detection __log:=/home/qyp/.ros/log/feb1051a-e38c-11eb-a2be-5378e1583e88/grasp_cloud_detection-2.log].
log file: /home/qyp/.ros/log/feb1051a-e38c-11eb-a2be-5378e1583e88/grasp_cloud_detection-2*.log

qyp-robot, 2021-07-13:
> @bostoncleek，你能帮助解决这个问题吗？
First of all, thank you very much for your reply, I still got the following error when I tested:
[grasp_cloud_detection-2] process has died [pid 411886, exit code -11, cmd /home/qyp/ws_grasp/devel/lib/moveit_task_constructor_gpd/grasp_cloud_detection __name:=grasp_cloud_detection __log:=/home/qyp/.ros/log/feb1051a -e38c-11eb-a2be-5378e1583e88/grasp_cloud_detection-2.log].
log file: /home/qyp/.ros/log/feb1051a-e38c-11eb-a2be-5378e1583e88/grasp_cloud_detection-2*.log

marcpujol-leitat, 2021-07-13:
Thank you @rhaschke for fast response! 

I have already merged the PR into my repo and now I am trying to modify the deep_grasp_demo package so it works with the changes.

I am having some troubles with that:
- Inside moveit_task_constructor_(gpd and dexnet) there are header files named grasp_detection.h which create a SimpleActionServer. I understand I need to modify those so they use the GraspPlanningAction. However, GraspPlanningFeedback and Result are missing members that the .cpp is using.
- In the commits of the PR I can see @bostoncleek has also created a class named GraspProvider. Should I be using that inside the deep_pick_and_place.cpp demo, for example in the Generate Grasp Pose stage.

Thanks again!

rhaschke, 2021-07-13:
I'm afraid I need to forward your questions to @bostoncleek. He implemented all this code originally.

qyp-robot, 2021-07-13:
Thank you very much for your reply. I am also working on it. If you have the latest progress, please let me know in time. Thank you very much!

bostoncleek, 2021-07-13:
You should use GraspProvider in both moveit_task_constructor_(gpd and dexnet) packages. 

You will also need to update the tutorial and demo to use grasping_msgs::GraspPlanningAction. If GraspPlanningFeedback and Result are missing make sure you include the grasping_msgs::GraspPlanningAction action message in you code. Check out the [action lib tutorials](http://wiki.ros.org/actionlib/Tutorials) for help. 

If you have more questions please let me know. I am trying to get permission from my employer to update the demo and tutorial. But if you finish the modifications rhaschke made above please submit a PR and I will review it. 

@marcpujol-leitat

marcpujol-leitat, 2021-07-13:
Hi,

I have created the PR so you can see the changes I've made.

Thank you!!

qyp-robot, 2021-07-14:
Hi,
I have changed the New grasp pose generator stage as per here: https://github.com/ros-planning/moveit_task_constructor/pull/196/commits/a2988995bcb85a099ca6766c709b90eac40d4432
This problem has been solved: header is missing

Second, I make changes according to here, Rename stage to Grasp Provider:
https://github.com/ros-planning/moveit_task_constructor/pull/196/commits/19fb72c81796303336d399905576a7de69602fc0
But there is a problem when compiling, what should I do, I look forward to your help!

@marcpujol-leitat
@bostoncleek.

qyp-robot, 2021-07-14:
Hello @bostoncleek,@marcpujol-leitat,@rhaschke 
When I change  according to here, Rename stage to Grasp Provider:
https://github.com/ros-planning/moveit_task_constructor/commit/19fb72c81796303336d399905576a7de69602fc0；

Do I also need to update the tutorial and demo to use grasping_msgs::GraspPlanningAction?
Then compile it?
Thank you all!

marcpujol-leitat, 2021-07-14:
Hi @qyp-robot,

Now that you have solved the header problems, I think you have to modify the deep_grasp_demo so it works with the changes made in moveit_task_constructor according to the links you have sent.

qyp-robot, 2021-07-14:
Thank you for your prompt reply!@marcpujol-leitat
I have followed your prompt:
https://github.com/PickNikRobotics/deep_grasp_demo/pull/11/commits/971d3db105473eb93ab948ff9cf675905df3f951

The demo to use grasping_msgs::GraspPlanningAction has been updated, but when I was compiling, an error occurred, such as:

rror: ‘const struct grasping_msgs::GraspPlanningFeedback_<std::allocator<void> >’ has no member named ‘grasp_candidates’
  175 | if (feedback->grasp_candidates.size() != feedback->costs.size()) {
      | ~~~~~~~~~~^~~~~~~~~~~~~~~~

error: ‘const struct grasping_msgs::GraspPlanningFeedback_<std::allocator<void> >’ has no member named ‘costs’
  175 | if (feedback->grasp_candidates.size() != feedback->costs.size()) {
      | ~~~~~~~~~~^~~~~

I checked carefully. In my /opt/ros/noetic/share/grasping_msgs/action, there is a file GraspPlanning.action. I don't know how to solve this compilation error. I hope you can help me! Thank you very much!

marcpujol-leitat, 2021-07-14:
During the changes you made in the moveit_task_constructor, you substituted the original action with GraspPlanningAction which no longer has "grasp_candidates" inside its feedback. You can find there message in the documentation: [Grasp Planning Action](http://docs.ros.org/en/indigo/api/grasping_msgs/html/action/GraspPlanning.html)
As bostoncleek already suggested, if you need help with that, you should check the [Action Lib Tutorials](http://wiki.ros.org/actionlib/Tutorials)

qyp-robot, 2021-07-14:
Thank you very much!@marcpujol-leitat
As you said, this is the problem, I am revising it, thank you! If I have any questions, I will ask you again. Thank you very much for your reply!

qyp-robot, 2021-07-14:
Hello, @marcpujol-leitat
when I changed the grass_provider, he originally used the information in SampleGraspPoses.action, the information is as follows:

# goal sent to client
string action_name
---
# result sent to server
string grasp_state
---
# feedback sent to server
# grasp poses
geometry_msgs/PoseStamped[] grasp_candidates
# cost of each grasp
float64[] costs

But the information in GraspPlanning.action is this:

# This action is used when planning grasps for a single,
# already known object, one object at a time.

# Object for which grasp planning is requested
Object object

# Name of group to plan with (optional)
string group_name
---
# All grasps
moveit_msgs/Grasp[] grasps
---
# Grasps found thus far
moveit_msgs/Grasp[] grasps

In grape_provider.h, I am not quite clear about what information in GraspPlanning.action should be used to replace the original costs message, and what should be its action_name; I’m not sure how to change it, can you help me? Thank you!

marcpujol-leitat, 2021-07-14:
Hi @qyp-robot,

If you go to my [Pull Request](https://github.com/PickNikRobotics/deep_grasp_demo/pull/11/files) you will see the changes I made to grasp_detection.cpp.
There you can see what I did to solve what you are mentioning.

However, **it is still not working and I am waiting for bostoncleek to review the changes**, so do not expect it to work propery.

qyp-robot, 2021-07-15:
Thank you very much for your reply.@marcpujol-leitat
 It was at night on our side before, and I didn’t have time to reply you in time. I'm sorry, I will go for the test, thank you very much!

qyp-robot, 2021-07-15:
Hello! @bostoncleek.  
Can you take a look at the changes of pr from @marcpujol-leitat：
https://github.com/PickNikRobotics/deep_grasp_demo/pull/11/files
Thank you very much!

qyp-robot, 2021-07-15:
Hi,@marcpujol-leitat
Have you solved the problem? The following message appears when I run the terminal, and it is stuck here:

[INFO] [1626318339.260666630]: Loading task parameters
[INFO] [1626318339.270120710]: Initializing task pipeline
[INFO] [1626318339.278097396]: Loading robot model'panda'...

I am also very anxious. I have checked almost all the information but have not resolved it. Do you have any directions? Thank you!

marcpujol-leitat, 2021-07-15:
Hi @qyp-robot,

This happens when you run moveit_task_constructor_demo demo.launch?

It is working for me. Did you follow the tutorials?

I am using Ubuntu 20.04 and ROS Noetic. In the same workspace I have the moveit_task_constructor I also have the moveit repository (noetic-devel) and panda_moveit_config.

Hope it helps.

qyp-robot, 2021-07-15:
Thank you for your reply!@marcpujol-leitat
The first thing I run is: roslaunch moveit_task_constructor_demo demo.launch
Then I ran the GPD package:
roslaunch moveit_task_constructor_gpd gpd_demo.launch

Then the terminal stays here all the time:
============= CLASSIFIER ======================
model_file:
weights_file: /home/qyp/shenduxuexi/gpd/models/lenet/15channels/params
batch_size: 1
=============================================
============ CANDIDATE FILTERING =============
candidate_workspace: -1.00 1.00 -1.00 1.00 -1.00 1.00
min_aperture: 0.0000
max_aperture: 0.8500
=============================================
============ CLUSTERING =====================
min_inliers: 1
=============================================

[INFO] [1626346266.686664813]: Loading task parameters
[INFO] [1626346266.700117082]: Initializing task pipeline
[INFO] [1626346266.703910978]: Loading robot model'panda'...

Moveit did not perform crawling, etc. I think there should be no problem with this gpd part, is there still a problem with the moveit_task_constructor package?

qyp-robot, 2021-07-15:
Hi@marcpujol-leitat
After I followed your changes before, it is here to change:
https://github.com/PickNikRobotics/deep_grasp_demo/pull/11/files

Later, when compiling, the terminal reported an error and could not compile. Later, I made changes according to this:
https://github.com/ros-planning/moveit_task_constructor/pull/196/commits/ae376b6fc703b75bcf41e6d6a46bba0c75b21311

Compilation is successful, and then there is a problem with the above operation

marcpujol-leitat, 2021-07-15:
In my case it runs past where you are stuck and gets stuck somewhere else.

I'm sorry but I don't know how to help you, the only thing I can tell you is to get bostoncleek's fork of moveit_task_constructor instead of the commit you are mentioning:

- [1] git clone https://github.com/ros-planning/moveit_task_constructor.git
- [2] git remote add pr-mtc https://github.com/bostoncleek/moveit_task_constructor
- [3] git fetch pr-mtc
- [4] git merge pr-mtc/pr-deep_grasp_stage

This way you will have the changes he made.

qyp-robot, 2021-07-15:
Hello,@marcpujol-leitat
Thanks for the tip, I checked this branch:
https://github.com/ros-planning/moveit_task_constructor/compare/master...bostoncleek:pr-deep_grasp_stage
He merged all the previous changes, there are a total of 8 file changes, right, all the changes made are here, right? I think it is the same as what you suggested! The same is true for my previous modification!

I branched his pr-deep_grasp_stage code:
https://github.com/bostoncleek/moveit_task_constructor/tree/pr-deep_grasp_stage

Copy it to my workspace. At the moment, my workspace has the following 2 function packages:
moveit_task_constructor
panda_moveit_config

Compiled and can run the following two launches normally:
roslaunch moveit_task_constructor_demo demo.launch
roslaunch moveit_task_constructor_demo pickplace.launch

So I think that the moveit_task_constructor function package and the basic repair are completed, and the problem may still be in the deep_grasp_demo function package. I will follow your prompts again to modify the code in the deep_grasp_demo function package.

Is my understanding right? Thank you for your continued help. Thank you very much!

marcpujol-leitat, 2021-07-15:
Yes, I think we should wait for bostoncleek to review our changes.

Well maybe @rhaschke could help with my issue, as it is a problem regarding the plan function of the class moveit::task_constructor::task

After running the moveit_task_constructor_demo demo.launch and the moveit_task_constructor_gpd gpd_demo.launch, it looks like it is able to create the grasps and select the best ones but gets stuck there, not executing the task.
The command window looks like this:
[image]

I have checked the code and it gets stuck here, in the task_.plan(max_solutions)
[image]

Where task_ is an object of the class moveit::task_constructor::task.

Do you have any idea of why this may be happening?

Thank you!

qyp-robot, 2021-07-15:
Sorry, I don't know how to solve it at the moment, I will try it, and if there is any progress, I must contact you in time. Thank you

qyp-robot, 2021-07-16:
Hello!@marcpujol-leitat
When I run launch:
roslaunch moveit_task_constructor_gpd gpd_demo.launch

An error occurred in the terminal, as shown below:

====> HAND SEARCH TIME: 0.340968
Generated 50 hand sets.
Filtering grasps outside of workspace ...
Number of grasp candidates within workspace and gripper width: 400
Number of grasp candidates with correct approach direction: 400
neighborhoods search time: 0.0258
Created 400 images in 1.1579s

[grasp_cloud_detection-2] process has died [pid 533124, exit code -11, cmd /home/qyp/ws_grasp/devel/lib/moveit_task_constructor_gpd/grasp_cloud_detection __name:=grasp_cloud_detection __log:=/home/qyp/.ros/log/5adf923e -e603-11eb-a2be-5378e1583e88/grasp_cloud_detection-2.log].
log file: /home/qyp/.ros/log/5adf923e-e603-11eb-a2be-5378e1583e88/grasp_cloud_detection-2*.log

The meaning here is that there is a problem with my point cloud search, but I have made the change according to your prompt:

grasp_cloud_detection.cpp
grasp_detection.h

These two files have been changed and the compilation is successful, but the above error still appears, it seems that an error occurred during the point cloud processing;

Is this a pcl problem? I follow this for pcl:
https://github.com/PickNikRobotics/deep_grasp_demo#Install
Run the script to install, he prompts:
PCL >= 1.9: The pcl_install.sh script will install PCL 1.11
Should I install the proper version of pcl to avoid this problem. Looking forward to your reply. Thank you very much!

qyp-robot, 2021-07-16:
My system configuration is the same as yours:
ubuntu 20.04
ros version: noetic
Can you help me, thank you very much!@marcpujol-leitat

marcpujol-leitat, 2021-07-16:
Hi @qyp-robot,

I can not really help much with that as you are not showing that much information... Maybe you should check the log that the error is pointing at, there you would find more information.

Regarding PCL, I builded version 1.11 after downloading https://github.com/PointCloudLibrary/pcl/archive/pcl-1.11.0.tar.gz 
Usually I do everything with apt-get install but in this case I builded everything (opencv, pcl, moveit) as I was having some troubles with compilation.

qyp-robot, 2021-07-16:
Hello,@marcpujol-leitat
I was looking at the rosout.log file, and the file finally displayed as follows:

1626423176.323402746 INFO /grasp_cloud_detection [/home/qyp/ws_grasp/src/deep_grasp_demo/moveit_task_constructor_gpd/src/grasp_detection.cpp:123(GraspDetection::goalCallback)] [topics: /genes,/grasrate, /genes /status] New goal accepted: object

Then there is no more. I guess that the operation of grab_detection.cpp has stopped here. I don’t know why. My pcl is also version 1.11. Everything else should be the same as you. I don’t know why it stopped. I hope you Can help me, thanks!

qyp-robot, 2021-07-16:
Hello!@marcpujol-leitat
My pcl version is 1.11,
opencv is 3.4
Eigen version is 3.3.7
But run:
roslaunch moveit_task_constructor_gpd gpd_demo.launch
Still there is a process has died. As mentioned earlier, I am not sure about the solution. Is the log file you mentioned earlier under .ros? I searched it but it seems that the information in it is useless, I hope you can help me, thank you!

qyp-robot, 2021-07-16:
Hello, sorry, I need to trouble you again!@marcpujol-leitat
When I run this launch:
roslaunch moveit_task_constructor_gpd gpd_demo.launch

The terminal has the following problems:
ERROR: Cannot open file: /home/qyp/shenduxuexi/gpd/models/lenet/15channels/paramsconv1_weights.bin!
ERROR: Cannot open file: /home/qyp/shenduxuexi/gpd/models/lenet/15channels/paramsconv1_biases.bin!

In fact, the path of the conv1_weights.bin file is:
/home/qyp/shenduxuexi/gpd/models/lenet/15channels/params/

But I can't find where to change this path, can you help me, thank you very much!

marcpujol-leitat, 2021-07-16:
Hi @qyp-robot

This is explained in the [deep_grasp_demo repository](https://github.com/PickNikRobotics/deep_grasp_demo).

Check point 5 of the GPD Installation.

qyp-robot, 2021-07-16:
Thank you,@marcpujol-leitat
I have basically solved the above problems. The reason for the process died is:
ERROR: Cannot open file: /home/qyp/shenduxuexi/gpd/models/lenet/15channels/paramsconv1_weights.bin!
ERROR: Cannot open file: /home/qyp/shenduxuexi/gpd/models/lenet/15channels/paramsconv1_biases.bin!

It has been resolved, but I seem to have the same problem as yours, that is, after passing the GPD, I must have obtained the 94 best crawls, but after the score evaluation, the terminal stays here, as shown below Show:

Grasp 89: 572.777
Grasp 90: 571.187
Grasp 91: 540.53
Grasp 92: 527.825
Grasp 93: 477.965
Selected the 94 best grasps.
======== RUNTIMES ========
  1. Candidate generation: 0.3187s
  2. Descriptor extraction: 1.1264s
  3. Classification: 7.6959s
==========
  TOTAL: 9.1559s

Hope to get your help, thank you very much for your continued help!
@marcpujol-leitat，@bostoncleek.，@rhaschke

marcpujol-leitat, 2021-07-16:
It looks like we have the same problem now... Maybe we need to further modify the moveit_task_constructor_gpd as the planning phase gets stuck somewhere. My understanding of the code is not deep enough to know what to do so I will wait for @bostoncleek to review my repo.

@qyp-robot if you manage to get it working or solve this problem, please keep me updated.

Thanks!

qyp-robot, 2021-07-20:
Hello,@marcpujol-leitat
I have successfully run in the latest attempt:
roslaunch moveit_task_constructor_gpd gpd_demo.launch

My approach is as follows:
1. I did not modify the GraspProvider. I modified the file submitted at the beginning. The first file was compiled with an error of missing header files. I made the changes as follows:
https://github.com/ros-planning/moveit_task_constructor/pull/196/commits/a2988995bcb85a099ca6766c709b90eac40d4432

2. Do not modify the files in deep_grasp_demo, use SampleGraspPoses.action directly. I think this is also the way @bostoncleek. was implemented at the beginning; I will test it in gazebo, and if all are successful, I will look back at your changes. Where should we go to improve it, hope this will help you, thank you!

qyp-robot, 2021-08-03:
Hello!@marcpujol-leitat
I found that in gazebo, my panda robot model keeps shaking. I see that its quality should be set by the author. I can’t think of how to adjust the inertia matrix and quality, or should I find a 3D model for measurement?
Can you help me, thank you very much!

Hi-Zed, 2021-08-04:
> Yes, I think we should wait for bostoncleek to review our changes.
>
> Well maybe @rhaschke could help with my issue, as it is a problem regarding the plan function of the class [...]

I solved this and managed to run the demo using GPD.

The issue is [https://github.com/bostoncleek/moveit_task_constructor/blob/ae376b6fc703b75bcf41e6d6a46bba0c75b21311/core/src/stages/grasp_provider.cpp#L97](here). In the implementation of the grasp provider.

At line 97 there is a request to lock a mutex, but the same mutex was already locked at line 161. I am not even sure why the mutex is needed, I removed both lines and the demo works like a charm.

bostoncleek, 2022-01-09:
The GPD demo works now. Update by pulling the latest changes:

1) https://github.com/bostoncleek/deep_grasp_demo/tree/update_generators
2) https://github.com/bostoncleek/moveit_task_constructor/tree/pr-deep_grasp_stage

mathieugouletcrvi, 2022-01-24:
Hi @bostoncleek I tried to build the demo with the new files from your branch and it seems that the file moveit/utils/moveit_error_code.h is not found in task_p.h . I saw this https://githubmemory.com/repo/ros-planning/moveit_task_constructor/issues/331 from @rhaschke . I tried to merge the branch on the master but I am not still not able to build. 

Is there is a way to find the file or to modify the dependency to build the demo ? 

Thank you.
_________________________________________________________________________________
Errors     << moveit_task_constructor_core:make /home/labcrvi/ws_grasp/logs/moveit_task_constructor_core/build.make.009.log
In file included from /home/labcrvi/ws_grasp/src/moveit_task_constructor/core/include/moveit/task_constructor/task_p.h:42:0,
                 from /home/labcrvi/ws_grasp/src/moveit_task_constructor/core/src/introspection.cpp:38:
/home/labcrvi/ws_grasp/src/moveit_task_constructor/core/include/moveit/task_constructor/task.h:50:10: fatal error: moveit/utils/moveit_error_code.h: No such file or directory
 #include <moveit/utils/moveit_error_code.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

bostoncleek, 2022-01-25:
@mathieugouletcrvi You should build the demo using the branch in [PR 14](https://github.com/PickNikRobotics/deep_grasp_demo/pull/14). This is currently working on my end. Please let me know if you experience issues.

mathieugouletcrvi, 2022-01-25:
@bostoncleek Hi, I restarted with a clean work space for both ws_moveit and ws_grasp and took the branch in PR 14 . 

git clone https://github.com/bostoncleek/moveit_task_constructor.git -b pr-deep_grasp_stage
git clone https://github.com/bostoncleek/deep_grasp_demo.git -b update_generators 

Unfortunatly, I still have a broken dependency from task_p.h on moveit/utils/moveit_error_code.h 

I am on ubuntu 18.04 and ROS melodic running on a virtualbox VM. Is there something else that I should try ?

Thank you

//--------------------------------------------------------------

labcrvi@ubuntu18-vm:~/ws_grasp$ catkin build
-----------------------------------------------------------
Profile:                     default
Extending:        [explicit] /home/labcrvi/ws_moveit/devel
Workspace:                   /home/labcrvi/ws_grasp
-----------------------------------------------------------
Build Space:        [exists] /home/labcrvi/ws_grasp/build
Devel Space:        [exists] /home/labcrvi/ws_grasp/devel
Install Space:      [unused] /home/labcrvi/ws_grasp/install
Log Space:          [exists] /home/labcrvi/ws_grasp/logs
Source Space:       [exists] /home/labcrvi/ws_grasp/src
DESTDIR:            [unused] None
-----------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-----------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------
Workspace configuration appears valid.
-----------------------------------------------------------
[build] Found '9' packages in 0.0 seconds.                                                                               
[build] Package table is up to date.                                                                                     
Starting  >>> moveit_task_constructor_msgs                                                                               
Starting  >>> rviz_marker_tools                                                                                          
Finished  <<< moveit_task_constructor_msgs                         [ 0.9 seconds ]                                       
Finished  <<< rviz_marker_tools                                    [ 1.3 seconds ]                                       
Starting  >>> moveit_task_constructor_core                                                                               
_________________________________________________________________________________________________________________________
Errors     << moveit_task_constructor_core:make /home/labcrvi/ws_grasp/logs/moveit_task_constructor_core/build.make.001.log
In file included from /home/labcrvi/ws_grasp/src/moveit_task_constructor/core/include/moveit/task_constructor/task_p.h:42:0,
                 from /home/labcrvi/ws_grasp/src/moveit_task_constructor/core/src/introspection.cpp:38:
/home/labcrvi/ws_grasp/src/moveit_task_constructor/core/include/moveit/task_constructor/task.h:50:10: fatal error: moveit/utils/moveit_error_code.h: No such file or directory
 #include <moveit/utils/moveit_error_code.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/CMakeFiles/moveit_task_constructor_core.dir/introspection.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [src/CMakeFiles/moveit_task_constructor_core.dir/all] Error 2
make: *** [all] Error 2
cd /home/labcrvi/ws_grasp/build/moveit_task_constructor_core; catkin build --get-env moveit_task_constructor_core | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
.........................................................................................................................
Failed     << moveit_task_constructor_core:make                    [ Exited with code 2 ]                                
Failed    <<< moveit_task_constructor_core                         [ 6.1 seconds ]                                       
Abandoned <<< moveit_task_constructor_dexnet                       [ Unrelated job failed ]                              
Abandoned <<< moveit_task_constructor_gpd                          [ Unrelated job failed ]                              
Abandoned <<< deep_grasp_task                                      [ Unrelated job failed ]                              
Abandoned <<< moveit_task_constructor_capabilities                 [ Unrelated job failed ]                              
Abandoned <<< moveit_task_constructor_demo                         [ Unrelated job failed ]                              
Abandoned <<< moveit_task_constructor_visualization                [ Unrelated job failed ]                              
[build] Summary: 2 of 9 packages succeeded.                                                                              
[build]   Ignored:   None.                                                                                               
[build]   Warnings:  None.                                                                                               
[build]   Abandoned: 6 packages were abandoned.                                                                          
[build]   Failed:    1 packages failed.                                                                                  
[build] Runtime: 7.5 seconds total.

mathieugouletcrvi, 2022-01-25:
@bostoncleek Hi, I restarted with a clean work space for both ws_moveit and ws_grasp and took the branch in 

https://github.com/bostoncleek/deep_grasp_demo/tree/update_generators
https://github.com/bostoncleek/moveit_task_constructor/tree/pr-deep_grasp_stage

Unfortunatly, I still have a broken dependency from task_p.h on moveit/utils/moveit_error_code.h

If I try to take the branch in PR 14 from moveit_task_constructor, there are other build errors. I copied the result at the end of my message.

I am on ubuntu 18.04 and ROS melodic running on a virtualbox VM. Is there something else that I should try ?

Thank you

//--------------------------------------------------------------
-----------------------------------------------------------
Profile:                     default
Extending:        [explicit] /home/labcrvi/ws_moveit/devel
Workspace:                   /home/labcrvi/ws_grasp
-----------------------------------------------------------
Build Space:        [exists] /home/labcrvi/ws_grasp/build
Devel Space:        [exists] /home/labcrvi/ws_grasp/devel
Install Space:      [unused] /home/labcrvi/ws_grasp/install
Log Space:          [exists] /home/labcrvi/ws_grasp/logs
Source Space:       [exists] /home/labcrvi/ws_grasp/src
DESTDIR:            [unused] None
-----------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-----------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------
Workspace configuration appears valid.
-----------------------------------------------------------
[build] Found '7' packages in 0.0 seconds.                                                              
[build] Updating package table.                                                                         
Starting  >>> moveit_task_constructor_msgs                                                              
Starting  >>> rviz_marker_tools                                                                         
Finished  <<< moveit_task_constructor_msgs                         [ 5.2 seconds ]                      
Finished  <<< rviz_marker_tools                                    [ 11.2 seconds ]                     
Starting  >>> moveit_task_constructor_core                                                              
________________________________________________________________________________________________________
Errors     << moveit_task_constructor_core:make /home/labcrvi/ws_grasp/logs/moveit_task_constructor_core/build.make.002.log
In file included from /home/labcrvi/ws_grasp/src/moveit_task_constructor/core/src/properties.cpp:39:0:
/home/labcrvi/ws_grasp/src/moveit_task_constructor/core/include/moveit/task_constructor/properties.h:211:33: error: ‘vector’ in namespace ‘std’ does not name a template type
  size_t countDefined(const std::vector<std::string>& list) const;
                                 ^~~~~~
/home/labcrvi/ws_grasp/src/moveit_task_constructor/core/include/moveit/task_constructor/properties.h:211:39: error: expected ‘,’ or ‘...’ before ‘<’ token
  size_t countDefined(const std::vector<std::string>& list) const;
                                       ^
/home/labcrvi/ws_grasp/src/moveit_task_constructor/core/src/properties.cpp:163:8: error: prototype for ‘size_t moveit::task_constructor::PropertyMap::countDefined(const std::vector<std::__cxx11::basic_string<char> >&) const’ does not match any in class ‘moveit::task_constructor::PropertyMap’
 size_t PropertyMap::countDefined(const std::vector<std::string> &list) const
        ^~~~~~~~~~~
In file included from /home/labcrvi/ws_grasp/src/moveit_task_constructor/core/src/properties.cpp:39:0:
/home/labcrvi/ws_grasp/src/moveit_task_constructor/core/include/moveit/task_constructor/properties.h:211:9: error: candidate is: size_t moveit::task_constructor::PropertyMap::countDefined(int) const
  size_t countDefined(const std::vector<std::string>& list) const;
         ^~~~~~~~~~~~
make[2]: *** [src/CMakeFiles/moveit_task_constructor_core.dir/properties.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/labcrvi/ws_grasp/src/moveit_task_constructor/core/include/moveit/task_constructor/storage.h:43:0,
                 from /home/labcrvi/ws_grasp/src/moveit_task_constructor/core/include/moveit/task_constructor/stage.h:44,
                 from /home/labcrvi/ws_grasp/src/moveit_task_constructor/core/include/moveit/task_constructor/container.h:41,
                 from /home/labcrvi/ws_grasp/src/moveit_task_constructor/core/include/moveit/task_constructor/container_p.h:41,
                 from /home/labcrvi/ws_grasp/src/moveit_task_constructor/core/src/container.cpp:37:
[... 34 lines cut ...]
                                                ^~~~~~~
/home/labcrvi/ws_grasp/src/moveit_task_constructor/core/src/marker_tools.cpp:67:42: error: template-id ‘element<>’ for ‘const VisualSharedPtr& moveit::task_constructor::element(const urdf::Link&)’ does not match any template declaration
 template <> const urdf::VisualSharedPtr& element(const urdf::Link& link) { return link.visual; }
[... cut ...]
[build] Summary: 2 of 7 packages succeeded.                                                             
[build]   Ignored:   None.                                                                              
[build]   Warnings:  None.                                                                              
[build]   Abandoned: 4 packages were abandoned.                                                         
[build]   Failed:    1 packages failed.                                                                 
[build] Runtime: 26.5 seconds total.

rhaschke, 2022-01-25:
@mathieugouletcrvi, did you update your MoveIt installation? We have cut a new release into MoveIt (1.0.9) recently, which is not yet synced. Consider using the debians from the [ROS `testing` repo](http://wiki.ros.org/TestingRepository).

mathieugouletcrvi, 2022-01-25:
@rhaschke Hi, I tried to use ROS testing repo. It still does not work... If I go with the master branch, I have error including SampleGraspPosesAction.h in grasp_image_detection.cpp and grasp_detection.h. The other error is still moveit_error_code.h missing from task.h.

@bostoncleek Hi, If I use the testing repo with the PR14 branch, I still miss moveit_error_code.h in task.h. I found the file here : https://docs.ros.org/en/api/moveit_core/html/cpp/moveit__error__code_8h_source.html and pasted it in the same directory as task.h, changing the path to #include "moveit_error_code.h" With this, I am able to build all the package, but with this message 

[build] An internal error occurred!                                                                                             
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/catkin_tools/verbs/catkin_build/build.py", line 549, in build_isolated_workspace
    continue_without_deps=False))
  File "/usr/lib/python2.7/dist-packages/catkin_tools/execution/executor.py", line 374, in run_until_complete
    return loop.run_until_complete(coroutine)
  File "/usr/lib/python2.7/dist-packages/trollius/base_events.py", line 350, in run_until_complete
    return future.result()
  File "/usr/lib/python2.7/dist-packages/trollius/futures.py", line 285, in result
    compat.reraise(type(self._exception), self._exception, exc_tb)
  File "/usr/lib/python2.7/dist-packages/trollius/tasks.py", line 259, in _step
    result = coro.send(value)
  File "/usr/lib/python2.7/dist-packages/trollius/tasks.py", line 387, in wait
    raise ValueError('Set of coroutines/Futures is empty.')
ValueError: Set of coroutines/Futures is empty.

If I still try to run the demo, roslaunch moveit_task_constructor_demo demo.launch launch rviz with the panda robot with thie error 

[ INFO] [1643136263.170738920]: Using planning interface 'OMPL'
[ERROR] [1643136263.176365833]: Exception while loading planning adapter plugin 'default_planner_request_adapters/ResolveConstraintFrames': According to the loaded plugin descriptions the class default_planner_request_adapters/ResolveConstraintFrames with base class type planning_request_adapter::PlanningRequestAdapter does not exist. Declared types are  default_planner_request_adapters/AddIterativeSplineParameterization default_planner_request_adapters/AddTimeOptimalParameterization default_planner_request_adapters/AddTimeParameterization default_planner_request_adapters/Empty default_planner_request_adapters/FixStartStateBounds default_planner_request_adapters/FixStartStateCollision default_planner_request_adapters/FixStartStatePathConstraints default_planner_request_adapters/FixWorkspaceBounds

If I try roslaunch moveit_task_constructor_gpd gpd_demo.launch . It does not work  and gives this error message 

ERROR: cannot launch node of type [deep_grasp_task/deep_grasp_demo]: Cannot locate node of type [deep_grasp_demo] in package [deep_grasp_task]. Make sure file exists in package path and permission is set to executable (chmod +x)
ERROR: cannot launch node of type [moveit_task_constructor_gpd/grasp_cloud_detection]: Cannot locate node of type [grasp_cloud_detection] in package [moveit_task_constructor_gpd]. Make sure file exists in package path and permission is set to executable (chmod +x)

It is the same for roslaunch moveit_task_constructor_dexnet dexnet_demo.launch

ERROR: cannot launch node of type [deep_grasp_task/deep_grasp_demo]: Cannot locate node of type [deep_grasp_demo] in package [deep_grasp_task]. Make sure file exists in package path and permission is set to executable (chmod +x)
ERROR: cannot launch node of type [moveit_task_constructor_dexnet/grasp_image_detection]: Cannot locate node of type [grasp_image_detection] in package [moveit_task_constructor_dexnet]. Make sure file exists in package path and permission is set to executable (chmod +x)

I will try on with noetic to see if I get the same result. 

Thank you.

qyp-robot, 2022-03-08:
Hello, @rhaschke @bostoncleek may I ask, I have completed the linkage of panda robot model between moveit and gazebo, but when I start gazebo and moveit; and then start roslaunch moveit_task_constructor_gpd gpd_demo.launch; the page is as follows. GPD has solved the result; but the robot in gazebo is not moving; I am not sure if this task classifier is not sending the trajectory to gazebo?

qyp-robot, 2022-03-08:
[image]

mathieugouletcrvi, 2022-03-08:
Hi @qyp-robot which branches have you used when you said : I have completed the linkage of panda robot model between moveit and gazebo.

Is it :
https://github.com/bostoncleek/deep_grasp_demo.git -b update_generators
https://github.com/bostoncleek/moveit_task_constructor.git -b pr-deep_grasp_stage

And
https://github.com/ros-planning/panda_moveit_config.git -b melodic-devel  
Or 
https://github.com/ros-planning/panda_moveit_config.git -b noetic-devel
And
https://github.com/tahsinkose/franka_ros.git -b melodic-devel  
Or  
https://github.com/frankaemika/franka_ros/tree/melodic-devel -b melodic-devel

Or have you used the setup assistant to generate the panda_moveit_config ?
Did you have to modify the files to make the robot follow rviz in gazebo ?

I opened this issue to discuss the topic :
https://github.com/PickNikRobotics/deep_grasp_demo/issues/15

The best I have done so far was to operate the panda robot in rviz only, with dexnet and gpd, but the robot is still falling under gravity in gazebo and does not moved. Any help would be greatly appreciated.

qyp-robot, 2022-03-08:
ok, i'll discuss it with you in time.@mathieugouletcrvi,

qyp-robot, 2022-03-10:
> Hello, @rhaschke @bostoncleek may I ask, I have completed the linkage of panda robot model between moveit and gazebo, but when I start gazebo and moveit; and then start roslaunch [...]

I have seen solutions to this problem elsewhere, thank you very much