moveit/moveit_task_constructor #282: roslaunch file contains multiple nodes named [/controller_spawner]

qyp-robot, 2021-07-13:
Hello everyone, when I run roslaunch deep_grasp_task gazebo_pick_place.launch, the terminal reports the following error:

RLException: roslaunch file contains multiple nodes named [/controller_spawner].
Please check all <node>'name' attributes to make sure they are unique.
Also check that $(anon id) use different ids.
The traceback for the exception was written to the log file

I searched in the lanuch file, but no duplicate node name was found. What should I do?
Looking forward to your answers, thank you very much!

v4hn, 2021-07-20:
> roslaunch deep_grasp_task gazebo_pick_place.launch

This issue is not in MTC itself, but `deep_grasp_task`.

closed 2021-07-20 by v4hn

qyp-robot, 2021-07-20:
Thank you, this problem has been solved, thank you very much!