moveit/moveit_task_constructor #285: Connect stages are causing error with ComputeIK wrapper.

Schteve-Bastian, 2021-08-17:
I am attempting to create task that plans from a serial container to an intermediate pose, and then to another container. Below is a code snippet with the two Connect stages and the IK wrapper. 

```c++
  {
    stages::Connect::GroupPlannerVector planner = { {parameters_.arm_group_name_, sampling_planner} };
    auto stage = std::make_unique<stages::Connect>("move to entry", planner);
    stage->properties().configureInitFrom(Stage::PARENT);
    t.add(std::move(stage));
  }

  {
    geometry_msgs::PoseStamped entry_pose;
    entry_pose.header.frame_id = "world";
    entry_pose.pose.position.x = 0.20;
    entry_pose.pose.position.y = -0.5;
    entry_pose.pose.position.z = 1.726;
    entry_pose.pose.orientation.x = 1.0;

    const Eigen::Isometry3d& hold_transform = current_scene.getFrameTransform(parameters_.hold_frame_);
    Eigen::Isometry3d frame_transform = hand_frame_transform.inverse() * hold_transform;
    frame_transform.rotate(Eigen::AngleAxisd(M_PI, Eigen::Vector3d::UnitX()));

    // Create Stage
    auto stage = std::make_unique<stages::GeneratePose>("generate entry pose");
    stage->setPose(entry_pose);
    stage->properties().set("marker_ns", "entry_pose");
    stage->setMonitoredStage(attach_object_stage);  // Hook into current state

    // Compute IK
    auto wrapper = std::make_unique<stages::ComputeIK>("entry pose IK", std::move(stage));
    wrapper->setMaxIKSolutions(2);
    wrapper->setMinSolutionDistance(1.0);
    wrapper->setProperty("group", parameters_.arm_group_name_);
    wrapper->setProperty("eef", parameters_.eef_name_);
    wrapper->setIKFrame(frame_transform, parameters_.hand_frame_);
    wrapper->properties().configureInitFrom(Stage::INTERFACE, { "target_pose" });
    t.add(std::move(wrapper));
  }

  {
    stages::Connect::GroupPlannerVector planner = { {parameters_.arm_group_name_, sampling_planner} };
    auto stage = std::make_unique<stages::Connect>("move to place", planner);
    stage->properties().configureInitFrom(Stage::PARENT);
    t.add(std::move(stage));
  }
```

The IK wrapper plans fine. Occasionally, the first connect stage will plan, but the second stage never does. Usually, however, the first stage will plan once, fail, and then the entire program will crash with this message:

```src/moveit_task_constructor/core/src/container.cpp:411: void moveit::task_constructor::SolutionCollector<dir>::traverse(const moveit::task_constructor::SolutionBase&, const moveit::task_constructor::InterfaceState::Priority&) [with moveit::task_constructor::Interface::Direction dir = (moveit::task_constructor::Interface::Direction)0]: Assertion `!successor->isFailure()' failed.```

What does this error mean, and what information am I missing in the stage initialization?

v4hn, 2021-08-18:
Could you could put this into a complete (and possibly minimal) example via `FixedState`?
It looks like a failure wrongly got accepted as part of a potential solution sequence.

Schteve-Bastian, 2021-08-18:
@v4hn Unfortunately, our pipeline uses moveit_cpp and is wrapped in so much supporting software that creating a reprex is unfeasible. I did create one that uses the panda arm and move_group, but in that situation the program executes as expected. I can pose the code if you're interested. In any case, you were correct about the reason for failure. The traverse function was unable to assert that the plan had not failed (I guess it helps to read error messages). I've created a working band-aid that seems to have fixed the problem.

Starting at [line 403](https://github.com/ros-planning/moveit_task_constructor/blob/74d33c4ec0bbdb17c02870dd8e29f04d27cd4fa6/core/src/container.cpp#L403) in container.cpp:
```c++
	void traverse(const SolutionBase& start, const InterfaceState::Priority& prio) {
		const InterfaceState::Solutions& next = trajectories<dir>(*state<dir>(start));
		if (next.empty()) {  // when reaching the end, add the trace to solutions
			assert(prio.depth() == trace.size());
			assert(prio.depth() <= max_depth);
			solutions.emplace_back(std::make_pair(trace, prio));
		} else {
			for (SolutionBase* successor : next) {
        if (successor->isFailure()) {
          ROS_ERROR_STREAM("I done broke it");
          continue;
        }
        assert(!successor->isFailure());  // We shouldn't have invalid solutions
				trace.push_back(successor);
				traverse(*successor, prio + InterfaceState::Priority(1, successor->cost()));
				trace.pop_back();
			}
		}
	}
```
I added an if statement that checks to see if the successor has failed before asserting that it hasn't, and then skips the rest of the loop if it can't continue. Does this help determine where the problem might be?