moveit/moveit_task_constructor #288: clicking "show available stages" leads to system crash

robotiste, 2021-09-04:
branch - "old-ros2"

'''
$ ros2 launch moveit_task_constructor_demo demo.launch.py

[ERROR] [rviz2-1]: process has died [pid 62761, exit code -11, cmd '/opt/ros/foxy/lib/rviz2/rviz2 -d /home/vikas/ws1/install/moveit_task_constructor_demo/share/moveit_task_constructor_demo/config/mtc.rviz --ros-args -r __node:=rviz2 --params-file /tmp/launch_params_sw7756fm --params-file /tmp/launch_params_nql6vamd --params-file /tmp/launch_params_45r_m4co --params-file /tmp/launch_params_q3fqs7l7'].
'''

<figure class="video_container">
  <iframe src="https://youtu.be/0rCFCZgFtJw="0" allowfullscreen="true"> </iframe>
</figure>

robotiste, 2021-09-04:
@v4hn please delete this issue if you can
I created it here by mistake :(

robotiste, 2021-09-04:
@v4hn please delete this issue
I accidentally opened it up here

this issue is related to @JafarAbdi repo's https://github.com/JafarAbdi/moveit_task_constructor/tree/old-ros2

v4hn, 2021-09-04:
Please have a look @JafarAbdi

JafarAbdi, 2021-09-07:
The old version didn't have a visualization manager, which is what happens when clicking on this button, this issue should be fixed with the latest ros2 branch, I'll update the branch to target all ros2 versions (foxy/galactic/rolling)

closed 2021-09-07 by JafarAbdi

robotiste, 2021-09-07:
@JafarAbdi I would love to contribute to the branch and get it up and running as soon as possible
Please let me know when can we connect.