moveit/moveit_task_constructor #289: Pick and place demo node is in infinite loop

robotiste, 2021-09-04:
## Environment
    
    Repo: https://github.com/JafarAbdi/moveit_task_constructor
    ROS Distro: Foxy
    OS Version: Ubuntu 20.04
    Source or Binary build? Source
    If source, which branch? old-ros2

## Expected behaviour
```
$ ros2 launch moveit_task_constructor_demo pickplace.launch.py 
```
It should create tasks and arm start following the tasks

## Actual behaviour
```
$ ros2 launch moveit_task_constructor_demo pickplace.launch.py 
```
```
[pick_place_demo-1] [INFO] [1630787195.875211857] [moveit_task_constructor_demo]: Init moveit_task_constructor_demo
[pick_place_demo-1] 2021-09-05 01:56:35.890 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7422: open_and_lock_file failed -> Function open_port_internal
[pick_place_demo-1] 2021-09-05 01:56:35.890 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7424: open_and_lock_file failed -> Function open_port_internal
[pick_place_demo-1] 2021-09-05 01:56:35.892 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7423: open_and_lock_file failed -> Function open_port_internal
[pick_place_demo-1] 2021-09-05 01:56:35.892 [RTPS_TRANSPORT_SHM Error] Failed init_port fastrtps_port7425: open_and_lock_file failed -> Function open_port_internal
[pick_place_demo-1] [INFO] [1630787195.917769812] [pick_place_task]: Loading task parameters
[pick_place_demo-1] [INFO] [1630787195.918995363] [pick_place_task]: Initializing task pipeline
[pick_place_demo-1] Parsing robot urdf xml string.
[pick_place_demo-1] [INFO] [1630787195.929121621] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00124889 seconds
[pick_place_demo-1] [INFO] [1630787195.929164292] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[pick_place_demo-1] Link panda_link1 had 1 children
[pick_place_demo-1] Link panda_link2 had 1 children
[pick_place_demo-1] Link panda_link3 had 1 children
[pick_place_demo-1] Link panda_link4 had 1 children
[pick_place_demo-1] Link panda_link5 had 1 children
[pick_place_demo-1] Link panda_link6 had 1 children
[pick_place_demo-1] Link panda_link7 had 1 children
[pick_place_demo-1] Link panda_link8 had 1 children
[pick_place_demo-1] Link panda_hand had 2 children
[pick_place_demo-1] Link panda_leftfinger had 0 children
[pick_place_demo-1] Link panda_rightfinger had 0 children
[pick_place_demo-1] [INFO] [1630787195.948305631] [pick_place_task]: Start searching for task solutions
[pick_place_demo-1] [INFO] [1630787195.950385202] [moveit.ros_planning.planning_pipeline]: No '~planning_plugin' parameter specified, but only 'ompl_interface/OMPLPlanner' planning plugin is available. Using that one.
[pick_place_demo-1] [INFO] [1630787195.976864536] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[pick_place_demo-1] [WARN] [1630787196.002198402] [moveit.ompl_planning.model_based_planning_context]: It looks like the planning volume was not specified.
[pick_place_demo-1] [INFO] [1630787196.002556772] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'hand' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[pick_place_demo-1] [INFO] [1630787196.002909152] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:220 - hand/hand: Starting planning with 1 states already in datastructure
[pick_place_demo-1] [INFO] [1630787196.018011618] [moveit_kinematic_constraints.kinematic_constraints]: Constraint satisfied:: Joint name: 'panda_finger_joint1', actual value: 0.034998, desired value: 0.035000, tolerance_above: 0.000010, tolerance_below: 0.000010
[pick_place_demo-1] [INFO] [1630787196.018063167] [moveit_kinematic_constraints.kinematic_constraints]: Constraint satisfied:: Joint name: 'panda_finger_joint2', actual value: 0.034998, desired value: 0.035000, tolerance_above: 0.000010, tolerance_below: 0.000010
[pick_place_demo-1] [INFO] [1630787196.018090141] [moveit_collision_detection_fcl.collision_common]: Found a contact between 'panda_rightfinger' (type 'Robot link') and 'panda_link5' (type 'Robot link'), which constitutes a collision. Contact information is not stored.
[pick_place_demo-1] [INFO] [1630787196.018100314] [moveit_collision_detection_fcl.collision_common]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[pick_place_demo-1] [ERROR] [1630787196.033165010] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:253 - hand/hand: Unable to sample any valid states for goal tree
[pick_place_demo-1] [INFO] [1630787196.033200448] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:354 - hand/hand: Created 1 states (1 start + 0 goal)
[pick_place_demo-1] [INFO] [1630787196.033214257] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/src/SimpleSetup.cpp:140 - No solution found after 0.030429 seconds
[pick_place_demo-1] [INFO] [1630787196.033294169] [moveit.ompl_planning.model_based_planning_context]: Unable to solve the planning problem
[pick_place_demo-1] [WARN] [1630787196.089164289] [moveit.ompl_planning.model_based_planning_context]: It looks like the planning volume was not specified.
[pick_place_demo-1] [INFO] [1630787196.089315851] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'hand' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[pick_place_demo-1] [INFO] [1630787196.089420125] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:220 - hand/hand: Starting planning with 1 states already in datastructure
[pick_place_demo-1] [INFO] [1630787196.100433724] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:354 - hand/hand: Created 7 states (2 start + 5 goal)
[pick_place_demo-1] [INFO] [1630787196.100455243] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/src/SimpleSetup.cpp:138 - Solution found in 0.011074 seconds
[pick_place_demo-1] [INFO] [1630787196.111176809] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/src/SimpleSetup.cpp:179 - SimpleSetup: Path simplification took 0.010694 seconds and changed from 6 to 2 states
[pick_place_demo-1] [WARN] [1630787196.119977230] [moveit.ompl_planning.model_based_planning_context]: It looks like the planning volume was not specified.
[pick_place_demo-1] [INFO] [1630787196.120124160] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'hand' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[pick_place_demo-1] [INFO] [1630787196.120290332] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:220 - hand/hand: Starting planning with 1 states already in datastructure
[pick_place_demo-1] [INFO] [1630787196.141498264] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/planners/rrt/src/RRTConnect.cpp:354 - hand/hand: Created 7 states (2 start + 5 goal)
[pick_place_demo-1] [INFO] [1630787196.141524059] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric/src/SimpleSetup.cpp:138 - Solution found in 0.021293 seconds
[pick_place_demo-1] [INFO] [1630787196.149513119] [ompl]: /tmp/binarydeb/ros-foxy-ompl-1.5.0/src/ompl/geometric

https://user-images.githubusercontent.com/22465401/132107453-cb493524-e807-4377-968a-8a7afa64d9a5.mp4

/src/SimpleSetup.cpp:179 - SimpleSetup: Path simplification took 0.007968 seconds and changed from 6 to 2 states

....................
```
It looks like, it is stuck in some sort of loop 
It keeps creating tasks again and again.

@JafarAbdi @v4hn  Any idea why this is happening?

https://user-images.githubusercontent.com/22465401/132107463-615617cf-dbd8-4362-9baf-4659bb061912.mp4

v4hn, 2021-09-04:
the behavior is expected. it doesn't recreate the task, but plans through all alternatives in a loop. as you can see in the rviz panel it fails to find a solution for "open hands". that's probably because the arm /hand are in collision, aren't they? maybe the robot does not spawn in the correct configuration on startup.

closed 2021-09-04 by v4hn

robotiste, 2021-09-06:
@v4hn Yes. It was due to the collision.