moveit/moveit_task_constructor #291: Motion Planning tasks introspection is wrong when using map as fixed frame

YannickRiou, 2021-09-14:
Hi,

There seems to be a problem when trying to review the planned task stages on robot that is localized into a map. 

[image]

As you can see on the above picture, I want to review the "grasp pose IK" stage, when  I click on it, another robot with it's environment appears at the origin of the map (in the foreground of the above picture). 
The actual robot position is at the background of the above picture.

It seems also to affects the view of the poses generated by moveit task constructor. 

The expected behavior is that the stage review should show the robot at it's actual localized position

Feel free to ask if something is unclear.

v4hn, 2021-09-14:
Could you try to provide a minimal example?
Do you see similar problems with other robot links, or is this only a problem with your map frame?
Does your moveit config specify a planar/floating virtual joint?

rhaschke, 2021-09-15:
I am able to reproduce the issue. It's a bug in MoveIt's PlanningSceneDisplay actually. I will file a PR.

PR #292 (2021-09-15): Update main scene node's pose on updates of robot's model frame
That's the same fix as https://github.com/ros-planning/moveit/pull/2876.
This fixes #291 in a naive way: 
If the map changes over time, I would actually expect the solution display to use the robot frame transform at the time of planning - not the most recent one as implemented here. However, this timestamp isn't conveyed with our messages yet.
@v4hn @YannickRiou any opinions?
closed 2021-09-15 by v4hn