moveit/moveit_task_constructor #308: Bug in pruning

JafarAbdi, 2021-11-15:
I have a very simple task, that fails to find a solution because it's not trying to connect certain pairs, I created a [simple test](https://github.com/JafarAbdi/moveit_task_constructor/tree/pr-pruning_bug) the trigger it, I have spent so much time trying to debug it without any hope :(

```
Current State
Connect
ComputeIK
MoveRelative
```

I'll keep posting my findings, I would really appreciate any help

JafarAbdi, 2021-11-16:
Commenting https://github.com/ros-planning/moveit_task_constructor/blob/master/core/src/container.cpp#L189-L190 resolve it, but that disables pruning :/

rhaschke, 2021-11-18:
Thanks for reporting. Looks like a pruned branch isn't re-enabled again. I will have a look.

v4hn, 2021-11-18:
Hey Robert, I've been talking to Jafar already and am looking into it. Sorry, I should have commented here earlier...

The culprit is that pruning through solutions generated by connect stages does not take into consideration alternative connect approaches and the *opposite* state (in this case from the first generator) should be pruned, but still needs to be scheduled for futher connect attempts.

I would like to address this together with some other changes required for correct pruning related to [those you asked me to split out from my fallbacks patches](https://github.com/ros-planning/moveit_task_constructor/pull/287#issuecomment-922040435).
My work in progress for this is [here](https://github.com/v4hn/moveit_task_constructor/tree/pr-prune-connect), but some key issues are not addressed there yet.

rhaschke, 2021-11-19:
Thanks for the hint. I looked into your WIP branch and I'm (still) not convinced. I figured out another reason for the failure. Give me some more time to file a corresponding WIP fix.

rhaschke, 2021-11-19:
@v4hn, I pushed my WIP here: https://github.com/ros-planning/moveit_task_constructor/pull/309. I believe, the failing pruning through a `Fallbacks` container, is probably due to the `Fallbacks` itself as a `SerialContainer` works.

JafarAbdi, 2022-03-14:
Fixed with https://github.com/ros-planning/moveit_task_constructor/pull/309

closed 2022-03-14 by JafarAbdi