moveit/moveit_task_constructor #312: Scheduling

rhaschke, 2021-11-26:


DaniGarciaLopez, 2023-05-20:
Any updates on this feature? As I understand, this would greatly reduce the planning time as it allows parallel planning in different threads, right? 

We are very interested in this feature as our planning time is already too long for our application. The main solution we have found is to precompute the task offline, as mentioned [here](https://github.com/ros-planning/moveit_task_constructor/issues/198), but sometimes it is necessary to do it in real time.

Are there any plans to implement it in the near future?

rhaschke, 2023-05-20:
Unfortunately, we didn't have time to start working on this and I don't foresee any change of that situation.
However, you are highly welcome to contribute to the project.