moveit/moveit_task_constructor #316: [ros2]: TODOs

JafarAbdi, 2021-11-26:
- [ ] Fix the change to RemoteTaskModel to have an internal node for the GetSolution client (not having it will cause the client to never be able to get a response causing RViz to freeze): [link](https://github.com/ros-planning/moveit_task_constructor/blob/98ced788c8fc3f6e6f7a697ac65a786c37a48f85/visualization/motion_planning_tasks/src/remote_task_model.cpp#L196)
- [ ] Fix the change to IntrospectionPrivate to have an internal node for each task: [link](https://github.com/ros-planning/moveit_task_constructor/blob/98ced78/core/src/introspection.cpp#L72-L99)
- [ ] Port downstream workspaces: [link](https://github.com/ros-planning/moveit_task_constructor/blob/98ced788c8fc3f6e6f7a697ac65a786c37a48f85/.github/workflows/ci.yaml#L44-L45)
- [ ] Port integration tests: [core/package.xml](https://github.com/ros-planning/moveit_task_constructor/blob/98ced788c8fc3f6e6f7a697ac65a786c37a48f85/core/package.xml#L24-L29) [core/test/CMakeLists.txt](https://github.com/ros-planning/moveit_task_constructor/blob/98ced788c8fc3f6e6f7a697ac65a786c37a48f85/core/test/CMakeLists.txt#L69-L82)
- [ ] Remove the addition of a random ID to prevent warnings about multiple publishers within the same node: 
  - [ ] [remote_task_model.cpp](https://github.com/ros-planning/moveit_task_constructor/blob/98ced788c8fc3f6e6f7a697ac65a786c37a48f85/visualization/motion_planning_tasks/src/remote_task_model.cpp#L192-L195)
  - [ ] [current_state.cpp](https://github.com/ros-planning/moveit_task_constructor/blob/98ced788c8fc3f6e6f7a697ac65a786c37a48f85/core/src/stages/current_state.cpp#L73-L75)
  - [ ] [task.cpp](https://github.com/ros-planning/moveit_task_constructor/blob/98ced788c8fc3f6e6f7a697ac65a786c37a48f85/core/src/task.cpp#L275-L279)
  - [ ] [introspection.cpp](https://github.com/ros-planning/moveit_task_constructor/blob/98ced788c8fc3f6e6f7a697ac65a786c37a48f85/core/src/introspection.cpp#L105-L106)
  - [ ] [task_panel.cpp](https://github.com/ros-planning/moveit_task_constructor/blob/98ced788c8fc3f6e6f7a697ac65a786c37a48f85/visualization/motion_planning_tasks/src/task_panel.cpp#L226-L228)
- [ ] Remove the check for `__has_include(<tf2_eigen/tf2_eigen.hpp>)` once galactic is no longer supported
- [ ] Port demo tests: [demo/CMakeLists.txt](https://github.com/ros-planning/moveit_task_constructor/blob/98ced788c8fc3f6e6f7a697ac65a786c37a48f85/demo/CMakeLists.txt#L64-L65)
- [ ] Add a version to [MOVEIT_HAS_OBJECT_POSE](https://github.com/ros-planning/moveit_task_constructor/blob/98ced788c8fc3f6e6f7a697ac65a786c37a48f85/core/include/moveit/task_constructor/moveit_compat.h#L48) rather than setting it to 0
- [ ] Move [test_move_to.cpp#L125](https://github.com/ros-planning/moveit_task_constructor/blob/98ced788c8fc3f6e6f7a697ac65a786c37a48f85/core/test/test_move_to.cpp#L125) after the else statement once https://github.com/ros-planning/moveit/commit/95aa8d0aa062b59f2f1f9a207a636b7f99652cbc is synced: [link](https://github.com/ros-planning/moveit_task_constructor/blob/ed45970/core/test/test_move_to.cpp#L120-L127)

lm2292, 2022-07-23:
@JafarAbdi can the ROS 2 port be used? I am using Foxy distro.

JafarAbdi, 2022-07-26:
Hi @lm2292, yeah it's usable, actually we've been using it for multiple projects without any problem, but unfortunately foxy was missing some important fixes (I forgot what they were) which why it's not supported.

Galactic, Humble, and Rolling are the currently supported distros