moveit/moveit_task_constructor #323: SolutionBase.fillMessage() creates an empty message

AndyZe, 2021-12-21:
I'm trying to parse the fields of a Solution message so I can execute them individually. I've noticed that publishing the message works fine:

```
  if (task_.plan()) {
    task_.introspection().publishSolution(*task_.solutions().front());
```

But the message produced this way is empty:

```
task_.solutions().front()->fillMessage(solution);
RCLCPP_INFO_STREAM(node->get_logger(), "Task ID: " << solution.task_id);
```

(Nothing is printed)

JafarAbdi, 2021-12-21:
`fillMessage` doesn't set the `task_id`, have you tried printing the whole message?

```cpp
task_.solutions().front()->fillMessage(solution);
RCLCPP_INFO_STREAM(node->get_logger(), "Solution: " << moveit_task_constructor_msgs::msg::to_yaml(solution));
```

Do you mind providing a small reproducible example?

AndyZe, 2021-12-21:
That does work. Thanks Jafar.

closed 2021-12-21 by AndyZe