moveit/moveit_task_constructor #325: Crash on calling introspectionId()

AndyZe, 2021-12-22:
```
open_fingers_stage_->setGroup(finger_group_);
open_fingers_stage_->setGoal(arm_prefix_ + "fingers_open");
task_.add(std::move(open_fingers_stage_));
RCLCPP_ERROR_STREAM(LOGGER, "Fingers open id: " << open_fingers_stage_->introspectionId());  // Crash here
```

Stack trace:

[ERROR] [mtc_demo_node-1]: process has died [pid 30956, exit code -11, cmd '/opt/colcon_ws/install/bro_manipulation_manager/lib/bro_manipulation_manager/mtc_demo_node --ros-args --params-file /tmp/launch_params_hnsr2_j0 --params-file /tmp/launch_params_vr1h9ke9 --params-file /tmp/launch_params_u_eb_jru --params-file /tmp/launch_params_41h_88id --params-file /tmp/launch_params_k61na_z_'].

(There is nothing else relevant)

I can probably make a PR to check for nullptr here. (I assume that's what's happening)

rhaschke, 2021-12-22:
I guess you try to access the member variable after moving it into the task?
Otherwise, I don't see a reason why this should fail.

AndyZe, 2021-12-22:
Nope, I double checked. Here's a grep of `open_fingers_stage_` to verify:

```
src/MTC_manager.cpp:    open_fingers_stage_->setGroup(finger_group_);
src/MTC_manager.cpp:    open_fingers_stage_->setGoal(arm_prefix_ + "fingers_open");
src/MTC_manager.cpp:    task_.add(std::move(open_fingers_stage_));
include/bro_manipulation_manager/MTC_manager.hpp:  std::unique_ptr<moveit::task_constructor::stages::MoveTo> open_fingers_stage_;
```

rhaschke, 2021-12-22:
Where is the initialization/creation of `open_fingers_stage_`? Did you cover all files?

AndyZe, 2021-12-22:
Geez, you are right. Initialization was missing. Thanks

closed 2021-12-22 by AndyZe