moveit/moveit_task_constructor #327: Costs increase quite a bit when planning from a hard-coded robot state

AndyZe, 2021-12-29:
We recently switched from starting from the current robot state:

`place_task_.add(std::make_unique<CurrentState>(PlaceStageEnum::PLACE_STAGE_MAP.at(PlaceStageEnum::CURRENT)));`

... to planning from a hard-coded robot state:

```
  // For demo purposes only, start the planning for the task as if we were already at
  // the expected location
  {
    auto scene = std::make_shared<planning_scene::PlanningScene>(pick_task_.getRobotModel());
    auto & state = scene->getCurrentStateNonConst();
    state.setToDefaultValues(state.getJointModelGroup(eef_group_), arm_prefix_ + "close");
    state.setToDefaultValues(
      state.getJointModelGroup(finger_group_), arm_prefix_ + "fingers_close");
    state.setToDefaultValues(state.getJointModelGroup(arm_group_), arm_prefix_ + "pre_place");
    std::vector<double> temp_box_position{-0.2, 0, 0.05, 0, 0, 0};
    state.setJointPositions("box_" + arm_prefix_ + "place_joint", temp_box_position);

    place_task_.setRobotModel(scene->getRobotModel());

    auto fixed = std::make_unique<moveit::task_constructor::stages::FixedState>(
      PlaceStageEnum::PLACE_STAGE_MAP.at(PlaceStageEnum::CURRENT));
    fixed->setState(scene);
    place_task_.add(std::move(fixed));
  }
```

The MTC plan stills works but I noticed a lot of the stage costs have grown from sub-100 to ~1200. It's curious that they are all around 1200. Here is a list of stages with cost around 1200:

> return_to_pre_place
> open_fingers
> release_eoat
> retreat

You can see from the names that some of these stages are very simple, such as opening the end-effector. Do you think this could be caused by a collision with the box position that is hard-coded? Got any other insights on how the cost is calculated?

rhaschke, 2021-12-29:
Costs are defined via `setCostTerm()` and are either zero or `PathLength` by default. Which concrete code you are referring to??

AndyZe, 2021-12-29:
Thanks. There's no way I can share this codebase.  :/

I wonder if the cost jumped because the robot is dual-arm and only the right arm position was initialized.

v4hn, 2021-12-29:
Robert answered your question on where the values are computed. The rest seems to be specific to your setup and not an actual upstream bug.

As a rule of thumb: look at the states in RViz. If your hardcoded state differs from the expected one (especially joint rotations of \tau might be relevant and cannot even be seen in the visualization...) If you see huge costs it might be because some of the trajectories are very long...

closed 2021-12-29 by v4hn