moveit/moveit_task_constructor #331: moveit_error_code.h is missing during install

EspenTeigen, 2022-01-05:
OS: Ubuntu 18.04
ROS version: Melodic

catkin build gives me this error 

`Errors     << moveit_task_constructor_core:make /home/espente/ROS/tiago/logs/moveit_task_constructor_core/build.make.000.log                                                                                                                                 
In file included from /home/espente/ROS/tiago/src/moveit_task_constructor/core/include/moveit/task_constructor/task_p.h:42:0,
                 from /home/espente/ROS/tiago/src/moveit_task_constructor/core/src/introspection.cpp:38:
/home/espente/ROS/tiago/src/moveit_task_constructor/core/include/moveit/task_constructor/task.h:50:10: fatal error: moveit/utils/moveit_error_code.h: No such file or directory
 #include `<moveit/utils/moveit_error_code.h>`

I have tried to reinstall moveit without any changes.

rhaschke, 2022-01-05:
This file was only recently added to MoveIt and not yet released. You need to build MTC against a MoveIt source build for now.
A new Melodic release of MoveIt is pending, but we are waiting for a few more PRs to be reviewed and merged before.

EspenTeigen, 2022-01-05:
Thank you for a really quick response :)

closed 2022-01-05 by EspenTeigen