moveit/moveit_task_constructor #334: Missing moveit_error_code.h

cs-99, 2022-01-18:
I recently tried building the moveit_task_constructor, which did not work since moveit/utils/moveit_error_code.h was missing. It seems like the file was moved to MoveIts core (see https://github.com/ros-planning/moveit/pull/3009) but is still included from moveit/utils in task.h. I simply fixed it manually and everything worked fine.

rhaschke, 2022-01-18:
I'm confused. The [PR #3009](https://github.com/ros-planning/moveit/pull/3009) you mentioned exactly moved `moveit_error_code.h` into the location `moveit/utils` you mentioned. However, as the latest MoveIt releases (into Melodic and Noetic) are not yet synced, an installation from binary .debs probably will still miss this file.
Please provide more detailed information about your versions and what exactly you changed manually.

cs-99, 2022-01-18:
I run Noetic. "However, as the latest MoveIt releases (into Melodic and Noetic) are not yet synced, an installation from binary .debs probably will still miss this file." should be the issue here. I just added the file to moveit/utils.

closed 2022-01-18 by cs-99