moveit/moveit_task_constructor #336: Compilation failing for core test

captain-yoshi, 2022-02-14:
Updated MTC to latest commit (prior 8 month old commit). Tried to compile against a clean workspace but still getting an error from `moveit_task_constructor_core` test. Using ROS Noetic. Disabling test for now to compile correctly.
```bash
Errors     << moveit_task_constructor_core:make /home/captain-yoshi/ws/ros/gsoc_ws/logs/moveit_task_constructor_core/build.make.031.log                                                                   
In file included from /usr/src/googletest/googlemock/include/gmock/gmock-actions.h:51,
                 from /usr/src/googletest/googlemock/include/gmock/gmock.h:59,
                 from /home/captain-yoshi/ws/ros/gsoc_ws/src/moveit_task_constructor/core/test/stage_mockups.h:11,
                 from /home/captain-yoshi/ws/ros/gsoc_ws/src/moveit_task_constructor/core/test/stage_mockups.cpp:4:
/usr/src/googletest/googlemock/include/gmock/internal/gmock-internal-utils.h:460:37: error: ‘IndexSequence’ has not been declared
  460 | auto ApplyImpl(F&& f, Tuple&& args, IndexSequence<Idx...>) -> decltype(
      |                                     ^~~~~~~~~~~~~
/usr/src/googletest/googlemock/include/gmock/internal/gmock-internal-utils.h:460:50: error: expected ‘,’ or ‘...’ before ‘<’ token
  460 | auto ApplyImpl(F&& f, Tuple&& args, IndexSequence<Idx...>) -> decltype(
      |                                                  ^
/usr/src/googletest/googlemock/include/gmock/internal/gmock-internal-utils.h:469:27: error: ‘MakeIndexSequence’ was not declared in this scope
  469 |                           MakeIndexSequence<std::tuple_size<Tuple>::value>())) {
      |                           ^~~~~~~~~~~~~~~~~

```
Libgtest info
```bash
# apt install libgtest-dev
libgtest-dev is already the newest version (1.10.0-2).
```

rhaschke, 2022-02-14:
This looks like a gtest incompatibility. Do you use any non-standard version?

captain-yoshi, 2022-02-16:
Not that I know of. Will try to pinpoint the cause later today and report back.

captain-yoshi, 2022-02-16:
Prior 61496fe, everything compiles fine. The only thing that was added is `gmock`. 
https://github.com/ros-planning/moveit_task_constructor/compare/61496fe..61399d6

System information:
```shell
$ dpkg -L google-mock
/.
/usr
/usr/share
/usr/share/doc
/usr/share/doc/google-mock
/usr/share/doc/google-mock/changelog.Debian.gz
/usr/share/doc/google-mock/copyright
/usr/src
/usr/src/gmock

$ dpkg -s google-mock | grep '^Version:'
Version: 1.10.0-2

$ ldconfig -p | grep libgmock*
# Empty
```
Should I install `libgmock-dev` instead of `google-mock` ?

rhaschke, 2022-02-16:
I have `libgmock-dev` 1.10.0-2 installed. Looks like this dependency is missing from package.xml?

captain-yoshi, 2022-02-17:
Seems like @v4hn figured out the culprit that caused the [issue](https://github.com/ros-industrial/ros_industrial_cmake_boilerplate/issues/54). 

I was using an old version of STOMP which had a bug in one of it's dependency. Finally got it to compile after updating Stomp and cleaned up the workspace.

> I have libgmock-dev 1.10.0-2 installed. 

Compiled w/wo `libgmock-dev` and still compiled fine. 

> Looks like this dependency is missing from package.xml?

I think recent version of `gtest` builds `gmock` by default which targets `gtest` so not necessary... But I'm still a novice when it comes to cmake.
```cmake
#Note that googlemock target already builds googletest
option(BUILD_GMOCK "Builds the googlemock subproject" ON)
option(INSTALL_GTEST "Enable installation of googletest. (Projects embedding googletest may want to turn this OFF.)" ON)
 
if(BUILD_GMOCK)
  add_subdirectory( googlemock )
else()
  add_subdirectory( googletest )
endif()
```

Thank you both !

closed 2022-02-17 by captain-yoshi

captain-yoshi, 2022-03-27:
**FYI** The library `libgmock-dev` is really needed. I triggered the same bug today (with an up to date Stomp package) when the library is missing.

It seems tricky to add dependencies to gmock from this [post](https://answers.ros.org/question/199680/catkin-doesnt-play-nice-with-google-mock/?answer=318629#post-id-318629).