moveit/moveit_task_constructor #342: Why the raw Stage pointers?

AndyZe, 2022-03-18:
I'm just curious, I see a lot of raw pointers in the demos, e.g.

https://github.com/ros-planning/moveit_task_constructor/blob/3a30e498bf081c83e6d206b9c150001cfbd67ad7/demo/src/pick_place_task.cpp#L182

I also had difficulty caching anything but a Stage* so I can use it to retrieve the stage_id:

`auto const stage_id = const_introspection.stageId(stage);`  // stage is a raw Stage*

Is that intentional? I suppose it's not a huge deal but could lead to some issues.

AndyZe, 2022-03-18:
If stages were stored internally as `shared_ptr` rather than `unique_ptr`, I would be able to cache stage pointers as `shared_ptrs` (I think).

rhaschke, 2022-03-18:
We intentionally decided for `unique_ptr` because stages shouldn't be "cached": It is strictly forbidden to re-use the very same stage in a different location of the pipeline!

rhaschke, 2022-03-18:
You can easily access stages within a container by their name:
https://github.com/ros-planning/moveit_task_constructor/blob/986d3c876620bde714076c8659bb17483086061f/core/include/moveit/task_constructor/container.h#L55

AndyZe, 2022-03-18:
^ great, that might be what I needed  :+1:

closed 2022-03-18 by AndyZe

AndyZe, 2022-03-18:
@rhaschke it looks like `findChild` only works in containers, huh? For example a SerialContainer. How would I access a stage outside the container?

I'll try putting my entire pick/place pipeline in a SerialContainer. I think that would work fine.

rhaschke, 2022-03-18:
> How would I access a stage outside the container?

What exactly do you mean? Stages always reside in a container, aren't they?
We don't yet have a method to recursively search for a given stage within a hierarchy of containers if that's what you are looking for. A PR for this is highly welcome. I had a syntax like `name A/name B/name C` in mind, i.e. separating names in the hierarchy via slashes. However, this limits possible names of stages to not include slashes... @v4hn and me didn't agree yet on the best solution for this.

AndyZe, 2022-03-18:
> What exactly do you mean? Stages always reside in a container, aren't they?

Well, the pick/place demo seems to have some stages that are not in a container:  https://github.com/ros-planning/moveit_task_constructor/blob/6d104e837e6c79b2578d1f13f89b519942c34659/demo/src/pick_place_task.cpp#L218

The container starts at L233.

My immediate problem is resolved. This is just an academic discussion for me, now.

rhaschke, 2022-03-18:
Each stage is part of a container. The task itself is a SerialContainer (by default) as well :wink: