moveit/moveit_task_constructor #344: ROS2: Container `findChild()` crashes

AndyZe, 2022-03-22:
This crashes:

  uint32_t const unique_stage_id =
      grasp_serial_container_->findChild(stage_name)->introspectionId();

`[ERROR] [mtc_demo_node-1]: process has died [pid 1403635, exit code -11, cmd '/home/andy/.../install/task_planning/lib/task_planning/mtc_demo_node --ros-args --params-file /tmp/launch_params_vc5xku0m --params-file /tmp/launch_params_azk5w6er --params-file /tmp/launch_params__k4cuqrx --params-file /tmp/launch_params_8pncj8rp --params-file /tmp/launch_params_ksynqtpn'].`

I double-checked that the stage has the right name and it is added to `grasp_serial_container_`.

```
  stage->setName(GetStageName(PickStage::CAMERA_POSE));
  grasp_serial_container_->insert(std::move(stage));
```

JafarAbdi, 2022-03-22:
Do you have a stack trace or a reproducible example?

AndyZe, 2022-03-22:
Backtrace:

```
  1  - ←   3 →   -  3 / pick
    1  - ←   3 →   -  3 / Grasp
      1  - ←   1 →   -  0 / Current robot state
      -  0 →   1 →   -  0 / move_to_camera_pose
      -  0 →   1 →   -  1 / Allow collisions between EE and box
      -  1 →   3 ←   3  - / Connector stage: current to grasp
      3  - ←   3 ←   0  - / Cartesian motion to approach the box
      0  - ←   3 →   -  0 / grasp pose IK
        1  - ←   1 →   -  1 / Generating a grasp pose
      -  0 →   3 →   -  0 / Attach box
      -  0 →   3 →   -  0 / Lift
      -  0 →   3 →   -  3 / Disallow box/ee collisions
Trying pimpl->children()
--Type <RET> for more, q to quit, c to continue without paging--backtrace

Thread 1 "mtc_demo_node" received signal SIGSEGV, Segmentation fault.
moveit::task_constructor::ContainerBase::numChildren (this=this@entry=0x0)
    at /home/andy/ws_lab0/src/moveit_task_constructor/core/src/container.cpp:253
253             return pimpl()->children().size();
```

AndyZe, 2022-03-22:
I'm working on it. It's a large client demo that I can't share easily.

rhaschke, 2022-03-22:
Then you should try to extract a minimal example.

AndyZe, 2022-03-22:
Jafar figured this out, like usual. I was calling `grasp_serial_container_->findChild()` after I had already "moved it" to the Task.

The way to access a named stage from the Task is:

`task_.stages()->findChild(stage_name)->introspectionId();`  (for example)

closed 2022-03-22 by AndyZe

rhaschke, 2022-03-22:
That's what I kind of expected... Good, that you found the issue.