moveit/moveit_task_constructor #346: Documentation: 2 CurrentStates are not allowed

AndyZe, 2022-03-23:
So, it seems that a Task cannot have 2 CurrentState-type stages. Is that correct? It seems worth mentioning in the header file, at least. I can make a documentation PR.

Code:

```
void PickPlan::MakeInitialStateStage() {
  auto stage =
      std::make_unique<CurrentState>(GetStageName(PickStage::INITIAL_STATE));
  task_.add(std::move(stage));
}

Stage* PickPlan::MakePreDropStateStage() {
  auto stage =
      std::make_unique<CurrentState>(GetStageName(PickStage::PRE_DROP_STATE));
  auto stage_ptr = stage.get();
  task_.add(std::move(stage));
  return stage_ptr;
}
```

Calling `MakeInitialStateStage()` followed by `MakePreDropStateStage()` causes this error. (It is probably not meaningful -- `fp` is a custom library we are using):

```
[mtc_demo_node-1] terminate called after throwing an instance of 'tl::bad_expected_access<fp::Error>'
[mtc_demo_node-1]   what():  Bad expected access
[ERROR] [mtc_demo_node-1]: process has died [pid 617722, exit code -6, cmd '/home/andy/.../install/task_planning/lib/task_planning/mtc_demo_node --ros-args --params-file /tmp/launch_params_itfp1zu9 --params-file /tmp/launch_params_idq9ohda --params-file /tmp/launch_params_9ibjv7n6 --params-file /tmp/launch_params_edtbbxsp --params-file /tmp/launch_params_zo3ew_ki'].
```

v4hn, 2022-03-23:
technically you are absolutely allowed to use the stage more than once. there's even code in the ros2 branch just for this case (instance - specific node names).

practically however, it's not very intelligent to do that. instead it makes more sense to monitor the spawned solution of the first current state and spawn it as well.

JafarAbdi, 2022-05-31:
@AndyZe Could you please provide the stack trace or a minimal reproducible example?