moveit/moveit_task_constructor #350: ros2 pickplace.launch.py demo plan failed

Jeisongarcia9713, 2022-04-02:
I am trying to make work the pick and place demo in ros2 galactic, the demo.launch.py is throwing without any error even though when  i run the pick and place launch and the plain fails but there is no too much detailed about the problem.

```ros2 launch moveit_task_constructor_demo pickplace.launch.py
```
[INFO] [launch]: All log files can be found below /home/jeison/.ros/log/2022-04-01-20-27-21-530037-jeison-134961
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [pick_place_demo-1]: process started with pid [134963]
[pick_place_demo-1] [INFO] [1648862842.432076827] [moveit_task_constructor_demo]: Loading task parameters
[pick_place_demo-1] [INFO] [1648862842.432894505] [moveit_task_constructor_demo]: Initializing task pipeline
[pick_place_demo-1] [INFO] [1648862842.465088839] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00555339 seconds
[pick_place_demo-1] [INFO] [1648862842.465139221] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
[pick_place_demo-1] Link panda_link1 had 1 children
[pick_place_demo-1] Link panda_link2 had 1 children
[pick_place_demo-1] Link panda_link3 had 1 children
[pick_place_demo-1] Link panda_link4 had 1 children
[pick_place_demo-1] Link panda_link5 had 1 children
[pick_place_demo-1] Link panda_link6 had 1 children
[pick_place_demo-1] Link panda_link7 had 1 children
[pick_place_demo-1] Link panda_link8 had 1 children
[pick_place_demo-1] Link panda_hand had 2 children
[pick_place_demo-1] Link panda_leftfinger had 0 children
[pick_place_demo-1] Link panda_rightfinger had 0 children
[pick_place_demo-1] [WARN] [1648862842.515616923] [moveit_task_constructor_demo]: Failed to find 'ompl.planning_plugin'. Attempting to load pipeline from old parameter structure. Please update your MoveIt config.
[pick_place_demo-1] [INFO] [1648862842.574433035] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
[pick_place_demo-1] [INFO] [1648862842.583946106] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.path_tolerance' was not set. Using default value: 0.100000
[pick_place_demo-1] [INFO] [1648862842.583984388] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.resample_dt' was not set. Using default value: 0.100000
[pick_place_demo-1] [INFO] [1648862842.583995721] [moveit_ros.add_time_optimal_parameterization]: Param 'move_group.min_angle_change' was not set. Using default value: 0.001000
[pick_place_demo-1] [INFO] [1648862842.584037821] [moveit_ros.fix_workspace_bounds]: Param 'move_group.default_workspace_bounds' was not set. Using default value: 10.000000
[pick_place_demo-1] [INFO] [1648862842.584077177] [moveit_ros.fix_start_state_bounds]: Param 'move_group.start_state_max_bounds_error' was set to 0.100000
[pick_place_demo-1] [INFO] [1648862842.584088208] [moveit_ros.fix_start_state_bounds]: Param 'move_group.start_state_max_dt' was not set. Using default value: 0.500000
[pick_place_demo-1] [INFO] [1648862842.584110657] [moveit_ros.fix_start_state_collision]: Param 'move_group.start_state_max_dt' was not set. Using default value: 0.500000
[pick_place_demo-1] [INFO] [1648862842.584122086] [moveit_ros.fix_start_state_collision]: Param 'move_group.jiggle_fraction' was not set. Using default value: 0.020000
[pick_place_demo-1] [INFO] [1648862842.584143874] [moveit_ros.fix_start_state_collision]: Param 'move_group.max_sampling_attempts' was not set. Using default value: 100
[pick_place_demo-1] [INFO] [1648862842.584166847] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Add Time Optimal Parameterization'
[pick_place_demo-1] [INFO] [1648862842.584178631] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Resolve constraint frames to robot links'
[pick_place_demo-1] [INFO] [1648862842.584185921] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Workspace Bounds'
[pick_place_demo-1] [INFO] [1648862842.584193421] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Bounds'
[pick_place_demo-1] [INFO] [1648862842.584200068] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State In Collision'
[pick_place_demo-1] [INFO] [1648862842.584206713] [moveit.ros_planning.planning_pipeline]: Using planning request adapter 'Fix Start State Path Constraints'
[pick_place_demo-1] [INFO] [1648862842.589229282] [moveit_task_constructor_demo]: Start searching for task solutions
[pick_place_demo-1] [INFO] [1648862842.821540624] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'hand' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[pick_place_demo-1]   0  - ←   0 →   -  0 / pick_place_task
[pick_place_demo-1]     1  - ←   1 →   -  0 / applicability test
[pick_place_demo-1]       1  - ←   1 →   -  1 / current state
[pick_place_demo-1]     -  0 →   1 →   -  1 / open hand
[pick_place_demo-1]     -  1 →   0 ←   0  - / move to pick
[pick_place_demo-1]     0  - ←   0 →   -  0 / pick object
[pick_place_demo-1]       0  - ←   0 ←   0  - / approach object
[pick_place_demo-1]       0  - ←   0 →   -  0 / grasp pose IK
[pick_place_demo-1]        25  - ←  25 →   - 25 / generate grasp pose
[pick_place_demo-1]       -  0 →   0 →   -  0 / allow collision (hand,object)
[pick_place_demo-1]       -  0 →   0 →   -  0 / close hand
[pick_place_demo-1]       -  0 →   0 →   -  0 / attach object
[pick_place_demo-1]       -  0 →   0 →   -  0 / allow collision (object,support)
[pick_place_demo-1]       -  0 →   0 →   -  0 / lift object
[pick_place_demo-1]       -  0 →   0 →   -  0 / forbid collision (object,surface)
[pick_place_demo-1]     -  0 →   0 ←   0  - / move to place
[pick_place_demo-1]     0  - ←   0 →   -  0 / place object
[pick_place_demo-1]       0  - ←   0 ←   0  - / lower object
[pick_place_demo-1]       0  - ←   0 →   -  0 / place pose IK
[pick_place_demo-1]         0  - ←   0 →   -  0 / generate place pose
[pick_place_demo-1]       -  0 →   0 →   -  0 / open hand
[pick_place_demo-1]       -  0 →   0 →   -  0 / forbid collision (hand,object)
[pick_place_demo-1]       -  0 →   0 →   -  0 / detach object
[pick_place_demo-1]       -  0 →   0 →   -  0 / retreat after place
[pick_place_demo-1]     -  0 →   0 →   -  0 / move home
[pick_place_demo-1] [INFO] [1648862844.284035805] [moveit_task_constructor_demo]: Planning failed
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[pick_place_demo-1] [INFO] [1648864843.185120820] [rclcpp]: signal_handler(signal_value=2)
[INFO] [pick_place_demo-1]: process has finished cleanly [pid 134963]

JeisonUR, 2022-04-05:
I was debugging and the problem is in the stage "grasp pose IK" and is exactly in the compute function from the class ComputeIK https://github.com/ros-planning/moveit_task_constructor/blob/ros2/core/src/stages/compute_ik.cpp#L380 because is never succeed

IgorUnlimited, 2022-04-12:
@JafarAbdi

sillkjc, 2022-04-19:
This looks very similar to a problem I ran into. If you visualise the target pose the translation component is not being set correctly to the YAML value:
```yaml
    grasp_frame_transform:
    - 0.0
    - 0.0
    - 0.1
    - 1.571
    - 0.785
    - 1.571
    ```
Modifying the stage to hard code in a translation component which is not in collision allowed the IK to compute and proceed.

IgorUnlimited, 2022-04-21:
@sillkjc 
Did you manage to finish the pick and place in your case?
Because after use your hardcode hack I get error on next step
`[pick_place_demo-1] [ERROR] [1650540406.167990950] [rosparam_shortcuts]: Missing parameter 'moveit_task_constructor_demo.execute'.
`

JafarAbdi, 2022-04-21:
I'll take a look at this next week, if I remember correctly it was more of a bug in moveit2 than MTC, I'll keep you posted as I work on this

closed 2022-05-25 by AndyZe