moveit/moveit_task_constructor #354: [ros2] How to specify a controller to use when executing a trajectory through the ExecuteTaskSolutionCapability ROS interface?

schornakj, 2022-04-14:
This is similar to this MoveIt2 issue I posted: https://github.com/ros-planning/moveit2/issues/1182

When the ExecuteTaskSolutionCapability [converts the solution message](https://github.com/ros-planning/moveit_task_constructor/blob/27672f8e2dd6ae690308357ac61e6771293084f0/capabilities/src/execute_task_solution_capability.cpp#L146-L189) into a series of `ExecutableTrajectory` objects, it does not have a way to set the names of the controllers that should be used to execute each trajectory segment. This results in the TrajectoryExecutionManager having to figure out on its own what controller to use for each set of joints, which ([as described in the linked issue](https://github.com/ros-planning/moveit2/issues/1182)) doesn't always work correctly if there are multiple controllers to choose from.

One solution is to add new fields to the MTC [Solution](https://github.com/ros-planning/moveit_task_constructor/blob/ros2/msgs/msg/Solution.msg) or [SubTrajectory](https://github.com/ros-planning/moveit_task_constructor/blob/ros2/msgs/msg/SubTrajectory.msg) messages to define what controllers should be used for each segment. I think this is an important supporting piece to [this suggestion](https://github.com/ros-planning/moveit2/issues/726#issuecomment-953614181) by @v4hn about adding additional properties to MTC Stages to set the controller to use during execution. This would be an API change, though, so I wanted to get feedback from others about the best path forward.

rhaschke, 2022-04-16:
There is already a pending PR for this: https://github.com/ros-planning/moveit_task_constructor/pull/123

schornakj, 2022-04-20:
Thank you for linking that! I submitted #355 as a ROS 2 version of #123, with some additional updates to add suggestions from #123's review.