moveit/moveit_task_constructor #359: Using MTC to Plan a Circular Path

swiz23, 2022-05-04:
I'm having issues trying to use the ros2 branch of MTC to plan a circular path around a non-TCP frame. I wrote up a question at ROS Answers describing the problem. Instead of copying and pasting it, I'm linking to it [here](https://answers.ros.org/question/400041/using-mtc-to-plan-a-circular-path/). In a nutshell though the two main questions I have are:
- How to use MTC to plan a circular path around a non-TCP frame? Specifying the frame_id in a goal message to be the non-TCP frame has no impact (circular path is still generated only around the TCP frame).
- While rotating around the non-TCP frame, how to keep the orientation of the TCP constant?

swiz23, 2022-08-26:
@JafarAbdi,

Here's a quick way to see the non-circular path currently generated for a non-TCP frame. I tested using the latest ros2 branch of MTC and main branch of moveit2 at the time of this writing. I did the following:

- Include the following code snippet in the panda.urdf file located [here](https://github.com/ros-planning/moveit_resources/blob/ros2/panda_description/urdf/panda.urdf). This will add a 'non-TCP frame' called panda_ee_link as a child of the panda_hand link (currently, the SRDF defines the TCP frame as panda_link8). The 'panda_ee_link' frame is the frame, about which, I'd like to generate a purely rotational motion.
```
    <link name="panda_ee_link" />
    <joint name="panda_ee_joint" type="fixed">
        <origin rpy="0 0 0" xyz="0 0 0.3" />
        <parent link="panda_hand" />
        <child link="panda_ee_link" />
        <axis xyz="0 0 0" />
    </joint>
```
- Next, delete the specified lines in the modular.cpp file [here](https://github.com/ros-planning/moveit_task_constructor/blob/ros2/demo/src/modular.cpp#L69-L87) and replace with the following code snippet. This shows the issue I'm seeing both with a MoveRelative stage and a MoveTo stage. However, I've seen the issue with a Connect / GeneratePose stage combo as well. Note that I'm using the setIKFrame function to set the target frame to panda_ee_link.
```
        {  // rotate about panda_ee_link using MoveRelative
                auto stage = std::make_unique<stages::MoveRelative>("rx +45°", cartesian);
                stage->properties().configureInitFrom(Stage::PARENT);
                geometry_msgs::msg::TwistStamped twist;
                twist.header.frame_id = "panda_ee_link";
                twist.twist.angular.x = M_PI / 4.;
                stage->setIKFrame("panda_ee_link");
                stage->setDirection(twist);
                c->insert(std::move(stage));
        }

        {  // rotate about panda_ee_link using MoveTo
                auto stage = std::make_unique<stages::MoveTo>("rx -90°", cartesian);
                stage->properties().configureInitFrom(Stage::PARENT);
                geometry_msgs::msg::PoseStamped pose;
                pose.header.frame_id = "panda_ee_link";
                pose.pose.orientation.x = -0.7071068;
                pose.pose.orientation.w = 0.7071068;
                stage->setIKFrame("panda_ee_link");
                stage->setGoal(pose);
                c->insert(std::move(stage));
        }
```

- After rebuilding, launch the demo by `ros2 launch moveit_task_constructor_demo demo.launch.py`.
Then launch the 'modular' script by typing `ros2 launch moveit_task_constructor_demo modular.launch.py`.

The planning should be successful, but if you look at the 'panda_ee_link' frame during the 'rx +45°' and 'rx -90°' motion, you'll see that instead of the `panda_ee_link` frame rotating in place, there is also a translational component (which I wouldn't have expected).

swiz23, 2022-09-05:
Thanks @rhaschke,

What would be the best way to test this? PR #380 is a ROS1 fix while PR #1547 is a ROS2 fix. I tried to go about it by using your moveit2 cartesian-interpolator branch, then merging the MTC fix-cartesian-interpolation branch into the ros2 branch of MTC, but ended up with merge conflicts.

rhaschke, 2022-09-05:
If I find time tomorrow, I will try the merge.

rhaschke, 2022-09-06:
@swiz23: See https://github.com/ros-planning/moveit_task_constructor/pull/382 for a ros2 fix. You will also need https://github.com/ros-planning/moveit2/pull/1547

swiz23, 2022-09-07:
Thanks, I'll try this out!

swiz23, 2022-10-14:
Finally got the time to try this out and it works as expected!
I'd like to test this out on the code for my particular project. Will update here.

swiz23, 2022-10-20:
Tested for my application successfully! I'll close this issue once the PR gets merged.

sea-bass, 2022-10-28:
I'm actually working on using the [Pilz motion planner](https://github.com/ros-planning/moveit2/tree/main/moveit_planners/pilz_industrial_motion_planner) as a planner within MTC that will give you a single circular arc trajectory, so you don't need a bunch of piecewise waypoints.

If this is of interest, I can keep this issue updated as I find time to flesh out my proof of concept.

rhaschke, 2022-10-31:
Of course, the Pilz planner is another option to generate circular motions. But that wasn't the original issue here.

swiz23, 2022-10-31:
@sea-bass, yep - that's definitely another option. But actually, in some cases (like the one here), you wouldn't need to specify a bunch of piece-wise way points. You would just specify a desired end-rotation around a desired frame, and that's it.

Looking forward to seeing #380 merged in soon :)

closed 2022-11-01 by rhaschke

swiz23, 2022-11-02:
Hey @rhaschke,

Thanks so much for merging that in! I assume that will be merged into the ros2 branch soon as well?

rhaschke, 2022-11-02:
Yes, I‘m working on it…

rhaschke, 2022-11-02:
@swiz23, I merged the master branch, including #380, into ros2.

PR #380 (2022-11-01): Fix Cartesian interpolation
This fixes #359. There were actually two major issues:
1. #304 broke Cartesian interpolation, wrongly calculating the target frame when rotations were involved (see videos below, were the endeffector should move and rotate about the green axis)
2. MoveIt itself didn't yet support Cartesian interpolation for non-link frames. To fix this, I filed https://github.com/ros-planning/moveit/pull/3197.

old:

https://user-images.githubusercontent.com/5376030/187067390-dd6e7af7-80e9-4b2a-8623-e3b844185e89.mp4

new:

https://user-images.githubusercontent.com/5376030/187067403- [... cut]