moveit/moveit_task_constructor #368: moveit_configs_utils is missing

samiulextreem, 2022-05-30:
I am trying to install moveit_task_constructor in my workspace of ros 2 foxy. But so far i have met with failure.
 I have run the following command 
`cd src`
`git clone https://github.com/ros-planning/moveit_task_constructor.git -b ros2`
`vcs import < moveit_task_constructor/.repos`
`cd ..`
`colcon build --packages-select moveit_task_constructor `

But it throws a bunch of error.

When i run `rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y` from src it reports missing dependency moveit_config_utils which i can not find anywhere. Can anybody help me explaining this problem?

samiulextreem, 2022-05-31:
Hi @JafarAbdi 

I cloned the galactic branch and run the rosdep. It does not throws any problem. But when I try to build the package, It did not build.
I also want to mention that running `vcs import < moveit_task_constructor/.repos` line, it clone rosparam_shortcuts package which also throws the following error.

JafarAbdi, 2022-05-31:
Can you try using https://github.com/PickNikRobotics/rosparam_shortcuts/tree/foxy rather than `ros2` branch and see if it resolves the issue for you?

samiulextreem, 2022-06-01:
Hi @JafarAbdi 
Now i have taken a look at ros 2 foxy doc, i have not  found anything related to moveit_test_constructor. Can you confirm that this package indeed made to run on foxy instead of humble?

JafarAbdi, 2022-06-01:
I'm sorry, after looking at it again, I found out that we didn't support foxy for MTC, the minimal supported version is galactic

closed 2022-06-01 by JafarAbdi