moveit/moveit_task_constructor #378: Terminal output for propagator stages shows no solutions propagated

stephanie-eng, 2022-08-16:
When planning a task with propagator stages, the terminal output doesn't match what I expected.

My understanding of the terminal output is that the first number in a given row indicates the number of solutions propagated backward, the second indicates the number of solutions generated, and the third indicates the number of solutions propagated forward. When planning a task with propagator stages, I often see this output from a propagator stage (propagating forward) for a successful plan: ` 0 →   1 →   -  0`

If my understanding above is correct, this would mean no stages were propagated backward (expected), one solution was generated (expected), but and no solutions were propagated forward (unexpected). I would expect for a successful plan with solutions propagating forward, that the third number would be 1 instead of 0.

As an example, here's the output from an example task with only propagator stages that successfully planned:

```
[my_node-1]   1  - ←   1 →   -  1 / PropagatorOnlyTask
[my_node-1]       1  - ←   1 →   -  0 / InitialState
[my_node-1]       -  0 →   1 →   -  0 / MoveTo
[my_node-1]       -  0 →   1 →   -  0 / ModifyPlanningScene
[my_node-1]       -  0 →   1 →   -  0 / ModifyPlanningScene
[my_node-1]       -  0 →   1 →   -  0 / ModifyPlanningScene
[my_node-1]       -  0 →   1 →   -  0 / MoveTo
[my_node-1]       -  0 →   1 →   -  0 / ModifyPlanningScene
[my_node-1]       -  0 →   1 →   -  0 / MoveTo
[my_node-1]       -  0 →   1 →   -  0 / ModifyPlanningScene
[my_node-1]       -  0 →   1 →   -  0 / ModifyPlanningScene
[my_node-1]       -  0 →   1 →   -  1 / MoveTo
```

rhaschke, 2022-08-16:
This is indeed weird behavior. I'm sure that worked in the past. Need to have a look...

rhaschke, 2022-08-16:
I checked this. The figures don't reflect the number of propagated states (as you expected), but actually the current number of states within the incoming/outgoing InterfaceState queues. As the propagator stages remove those states after processing (in contrast to a connect stage), the numbers are zero there. Actually, this output was intended for debugging of state propagation and is not really intended as an end-user interface.

closed 2022-08-16 by rhaschke