moveit/moveit_task_constructor #384: Use a task's solution planning scene in future tasks

JSchaaf2, 2022-10-18:
I want to plan tasks ahead of time to avoid having to wait for the planning itself between executing tasks. My idea is that it should be possible to plan the next task as soon as there is a solution for the current task, if the plan is made using the planningscene at the end of the current task. This is mainly to deal with objects that the robot is moving in its current task.

There is the fixed_state stage which, if I understand it correctly, makes it possible to specify what the planning scene is at that point in the task.
But is it possible to get the planningscene from a task which has a solution, or use it or it's solutionmsg to construct a planning scene? I noticed the task interfaces has a sceneptr, but don't think that should be used, and am not sure if using the scene information the soluitonmsg contains is a good idea.

Thanks in advance and apologies if it is a duplicate of  #379.

rhaschke, 2022-10-20:
> My idea is that it should be possible to plan the next task as soon as there is a solution for the current task, if the plan is made using the PlanningScene at the end of the current task.

Yes, that's possible. However, ideally, you would just append the latter task to the former and plan them simultaneously. That's what MTC is about!

> There is the `FixedState` stage, which makes it possible to specify what the planning scene is at that point in the task.

Exactly. This stage is initialized with a `PlanningScenePtr`.

> I noticed the task interface has a scene pointer.

Not anymore. Where did you see this pointer?

> Apologies if it is a duplicate of #379.

Indeed the approach to access the Scene is very similar:
`task_->solutions()->front()->end()->scene()`

JSchaaf2, 2022-10-27:
> Yes, that's possible. However, ideally, you would just append the latter task to the former and plan them simultaneously. That's what MTC is about!

I am not sure that is practical in our case, as we might not know what the second task will be when the first one starts. I assumed it was not possible to add stages to a task that is already executing. But I don't (yet) see a big difference between adding another task or adding stages to the old one. At least when splitting up the solution messages to execute other code in-between as suggested in #162. 

> > I noticed the task interface has a scene pointer.

> Not anymore. Where did you see this pointer?

Apparently I did misremember, it was the `task_->solutions()->front()->end()->scene()`. It turns out I was doing something wrong causing it to always be a nullpointer. 
Thank you for your help.

closed 2022-10-27 by JSchaaf2

rhaschke, 2022-10-27:
> I assumed it was not possible to add stages to a task that is already executing.

That's true. It's not possible.