moveit/moveit_task_constructor #385: Building using catkin_make failed in ros-melodic

patwaririshab, 2022-10-19:
Description:
1. Was trying to catkin_make to build package on ros-melodic on nvidia jetpack (Ubuntu 16.04)
2. Errors:

- error: use of ‘auto’ in lambda parameter declaration only available with -std=c++14 or -std=gnu++14

Error Logs
```
/home/hivebotics/Developer/abluo_ros_ws/src/moveit_task_constructor/visualization/motion_planning_tasks/utils/flat_merge_proxy_model.cpp: In member function ‘std::vector<moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::ModelData>::iterator moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::find(const QObject*)’:
/home/hivebotics/Developer/abluo_ros_ws/src/moveit_task_constructor/visualization/motion_planning_tasks/utils/flat_merge_proxy_model.cpp:114:65: error: use of ‘auto’ in lambda parameter declaration only available with -std=c++14 or -std=gnu++14
   return std::find_if(data_.begin(), data_.end(), [model](const auto& data) { return data.model_ == model; });
                                                                 ^~~~
/home/hivebotics/Developer/abluo_ros_ws/src/moveit_task_constructor/visualization/motion_planning_tasks/utils/flat_merge_proxy_model.cpp: In lambda function:
/home/hivebotics/Developer/abluo_ros_ws/src/moveit_task_constructor/visualization/motion_planning_tasks/utils/flat_merge_proxy_model.cpp:114:91: error: request for member ‘model_’ in ‘data’, which is of non-class type ‘const int’
   return std::find_if(data_.begin(), data_.end(), [model](const auto& data) { return data.model_ == model; });
                                                                                           ^~~~~~
/home/hivebotics/Developer/abluo_ros_ws/src/moveit_task_constructor/visualization/motion_planning_tasks/utils/flat_merge_proxy_model.cpp: In member function ‘int moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::accumulatedRowCount(std::vector<moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::ModelData>::const_iterator, std::vector<moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::ModelData>::const_iterator) const’:
/home/hivebotics/Developer/abluo_ros_ws/src/moveit_task_constructor/visualization/motion_planning_tasks/utils/flat_merge_proxy_model.cpp:119:59: error: use of ‘auto’ in lambda parameter declaration only available with -std=c++14 or -std=gnu++14
   return std::accumulate(start, end, 0, [](int acc, const auto& d) { return acc + d.model_->rowCount(); });
                                                           ^~~~
/home/hivebotics/Developer/abluo_ros_ws/src/moveit_task_constructor/visualization/motion_planning_tasks/utils/flat_merge_proxy_model.cpp: In lambda function:
/home/hivebotics/Developer/abluo_ros_ws/src/moveit_task_constructor/visualization/motion_planning_tasks/utils/flat_merge_proxy_model.cpp:119:85: error: request for member ‘model_’ in ‘d’, which is of non-class type ‘const int’
   return std::accumulate(start, end, 0, [](int acc, const auto& d) { return acc + d.model_->rowCount(); });
                                                                                     ^~~~~~
In file included from /usr/include/c++/7/numeric:62:0,
                 from /usr/include/aarch64-linux-gnu/qt5/QtCore/qhashfunctions.h:47,
                 from /usr/include/aarch64-linux-gnu/qt5/QtCore/qlist.h:47,
                 from /usr/include/aarch64-linux-gnu/qt5/QtCore/qvariant.h:45,
                 from /usr/include/aarch64-linux-gnu/qt5/QtCore/qabstractitemmodel.h:43,
                 from /usr/include/aarch64-linux-gnu/qt5/QtCore/QAbstractItemModel:1,
                 from /home/hivebotics/Developer/abluo_ros_ws/src/moveit_task_constructor/visualization/motion_planning_tasks/utils/flat_merge_proxy_model.h:39,
                 from /home/hivebotics/Developer/abluo_ros_ws/src/moveit_task_constructor/visualization/motion_planning_tasks/utils/flat_merge_proxy_model.cpp:37:
/usr/include/c++/7/bits/stl_numeric.h: In instantiation of ‘_Tp std::accumulate(_InputIterator, _InputIterator, _Tp, _BinaryOperation) [with _InputIterator = __gnu_cxx::__normal_iterator<const moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::ModelData*, std::vector<moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::ModelData> >; _Tp = int; _BinaryOperation = moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::accumulatedRowCount(std::vector<moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::ModelData>::const_iterator, std::vector<moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::ModelData>::const_iterator) const::<lambda(int, const int&)>]’:
/home/hivebotics/Developer/abluo_ros_ws/src/moveit_task_constructor/visualization/motion_planning_tasks/utils/flat_merge_proxy_model.cpp:119:106:   required from here
/usr/include/c++/7/bits/stl_numeric.h:154:22: error: no match for call to ‘(moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::accumulatedRowCount(std::vector<moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::ModelData>::const_iterator, std::vector<moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::ModelData>::const_iterator) const::<lambda(int, const int&)>) (int&, const moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::ModelData&)’
  __init = __binary_op(__init, *__first);
           ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~
/usr/include/c++/7/bits/stl_numeric.h:154:22: note: candidate: void (*)(int, const int&) <conversion>
/usr/include/c++/7/bits/stl_numeric.h:154:22: note:   candidate expects 3 arguments, 3 provided
/home/hivebotics/Developer/abluo_ros_ws/src/moveit_task_constructor/visualization/motion_planning_tasks/utils/flat_merge_proxy_model.cpp:119:66: note: candidate: moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::accumulatedRowCount(std::vector<moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::ModelData>::const_iterator, std::vector<moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::ModelData>::const_iterator) const::<lambda(int, const int&)>
   return std::accumulate(start, end, 0, [](int acc, const auto& d) { return acc + d.model_->rowCount(); });
                                                                  ^
/home/hivebotics/Developer/abluo_ros_ws/src/moveit_task_constructor/visualization/motion_planning_tasks/utils/flat_merge_proxy_model.cpp:119:66: note:   no known conversion for argument 2 from ‘const moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::ModelData’ to ‘const int&’
In file included from /usr/include/c++/7/bits/stl_algobase.h:71:0,
                 from /usr/include/c++/7/algorithm:61,
                 from /usr/include/aarch64-linux-gnu/qt5/QtCore/qglobal.h:109,
                 from /usr/include/aarch64-linux-gnu/qt5/QtCore/qatomic.h:41,
                 from /usr/include/aarch64-linux-gnu/qt5/QtCore/qvariant.h:43,
                 from /usr/include/aarch64-linux-gnu/qt5/QtCore/qabstractitemmodel.h:43,
                 from /usr/include/aarch64-linux-gnu/qt5/QtCore/QAbstractItemModel:1,
                 from /home/hivebotics/Developer/abluo_ros_ws/src/moveit_task_constructor/visualization/motion_planning_tasks/utils/flat_merge_proxy_model.h:39,
                 from /home/hivebotics/Developer/abluo_ros_ws/src/moveit_task_constructor/visualization/motion_planning_tasks/utils/flat_merge_proxy_model.cpp:37:
/usr/include/c++/7/bits/predefined_ops.h: In instantiation of ‘bool __gnu_cxx::__ops::_Iter_pred<_Predicate>::operator()(_Iterator) [with _Iterator = __gnu_cxx::__normal_iterator<moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::ModelData*, std::vector<moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::ModelData> >; _Predicate = moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::find(const QObject*)::<lambda(const int&)>]’:
/usr/include/c++/7/bits/stl_algo.h:120:14:   required from ‘_RandomAccessIterator std::__find_if(_RandomAccessIterator, _RandomAccessIterator, _Predicate, std::random_access_iterator_tag) [with _RandomAccessIterator = __gnu_cxx::__normal_iterator<moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::ModelData*, std::vector<moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::ModelData> >; _Predicate = __gnu_cxx::__ops::_Iter_pred<moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::find(const QObject*)::<lambda(const int&)> >]’
/usr/include/c++/7/bits/stl_algo.h:161:23:   required from ‘_Iterator std::__find_if(_Iterator, _Iterator, _Predicate) [with _Iterator = __gnu_cxx::__normal_iterator<moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::ModelData*, std::vector<moveit_rviz_plugin::utils::FlatMergeProxyModelPrivate::ModelData> >; _Predicate = __gnu_cxx::__ops::_Ite [... cut ...]
moveit_task_constructor/visualization/motion_planning_tasks/utils/CMakeFiles/motion_planning_tasks_utils.dir/build.make:62: recipe for target 'moveit_task_constructor/visualization/motion_planning_tasks/utils/CMakeFiles/motion_planning_tasks_utils.dir/flat_merge_proxy_model.cpp.o' failed
make[2]: *** [moveit_task_constructor/visualization/motion_planning_tasks/utils/CMakeFiles/motion_planning_tasks_utils.dir/flat_merge_proxy_model.cpp.o] Error 1
CMakeFiles/Makefile2:11071: recipe for target 'moveit_task_constructor/visualization/motion_planning_tasks/utils/CMakeFiles/motion_planning_tasks_utils.dir/all' failed
```

rhaschke, 2022-10-19:
ROS Melodic targets Ubuntu 18.04 (not 16.04) and thus assumes a C++14 compiler by default.
So please ensure that your system compiler handles C++14.
https://gist.github.com/ax3l/53db9fa8a4f4c21ecc5c4100c0d93c94
https://unix.stackexchange.com/questions/309254/g-how-to-use-std-c14-be-default

patwaririshab, 2022-10-20:
Apologies, made an error, I am actually using Ubuntu 18.04 LTS [linux version 4.9.253-tegra]

I tried the steps above, however it did not resolve the issue.
I suspect it could be due to QT. I am using qt5.95 (default QT version)