moveit/moveit_task_constructor #391: Member mismatch in moveit_task_constructor_visualization build failure

quarkytale, 2022-11-09:
### Description
Build failing with `moveit2_tutorials` and its dependencies  on humble/focal while trying to build from source. Seems like a version mismatch but I can't figure out which package/commit this could be related to. An older issue mentioned `srdfdom` to be on the ${ROS_DISTRO} branch but I have that already on `release/humble/srdfdom/2.0.4-1`. Any suggestions where to look?

### Environment
* ROS Distro: Humble
* OS Version: Ubuntu 20.04
* Source build: `moveit_core` and `moveit2_tutorials` on `humble`, other deps generated from `rosinstall_generator` or on `ros2` branch

### Console output

```
[Processing: moveit_setup_app_plugins, moveit_setup_controllers, moveit_setup_core_plugins, moveit_setup_srdf_plugins, moveit_task_constructor_visualization]    
--- stderr: moveit_task_constructor_visualization                                                                                                                
/home/user/moveit2/src/moveit_task_constructor/visualization/motion_planning_tasks/src/task_panel.cpp: In constructor ‘moveit_rviz_plugin::TaskPanel::TaskPanel(QWidget*)’:
/home/user/moveit2/src/moveit_task_constructor/visualization/motion_planning_tasks/src/task_panel.cpp:94:89: error: ‘idClicked’ is not a member of ‘QButtonGroup’
   94 |  connect(d->tool_buttons_group, static_cast<void (QButtonGroup::*)(int)>(&QButtonGroup::idClicked), d->stackedWidget,
      |                                                                                         ^~~~~~~~~
make[2]: *** [motion_planning_tasks/src/CMakeFiles/motion_planning_tasks_rviz_plugin.dir/build.make:245: motion_planning_tasks/src/CMakeFiles/motion_planning_tasks_rviz_plugin.dir/task_panel.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:450: motion_planning_tasks/src/CMakeFiles/motion_planning_tasks_rviz_plugin.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_task_constructor_visualization [3min 30s, exited with code 2]
Aborted  <<< moveit_setup_app_plugins [2min 49s]                                                                                                                
Aborted  <<< moveit_setup_core_plugins [3min 19s]                                                                                                                
Aborted  <<< moveit_setup_controllers [5min 27s]                                                                                                             
Aborted  <<< moveit_setup_srdf_plugins [6min 25s]                                              

Summary: 117 packages finished [57min 33s]
  1 package failed: moveit_task_constructor_visualization
  4 packages aborted: moveit_setup_app_plugins moveit_setup_controllers moveit_setup_core_plugins moveit_setup_srdf_plugins
  42 packages had stderr output: angles controller_manager eigen_stl_containers generate_parameter_library generate_parameter_library_py geometric_shapes ignition_cmake2_vendor ignition_math6_vendor image_common joint_state_publisher joint_state_publisher_gui launch_param_builder libcurl_vendor moveit_common moveit_configs_utils moveit_planners moveit_plugins moveit_resources moveit_resources_fanuc_description moveit_resources_fanuc_moveit_config moveit_resources_panda_description moveit_resources_panda_moveit_config moveit_resources_pr2_description moveit_resources_prbt_moveit_config moveit_resources_prbt_pg70_support moveit_resources_prbt_support moveit_ros moveit_runtime moveit_setup_srdf_plugins moveit_task_constructor_visualization octomap python_qt_binding ros2_control ros2_control_test_assets ros2_controllers ros2controlcli rviz_assimp_vendor rviz_ogre_vendor urdfdom_py vision_opencv xacro yaml_cpp_vendor
  3 packages not processed
```

rhaschke, 2022-11-09:
With latest updates I had to adapt for Qt 5.15 running on Ubuntu Jammy. To maintain compatibility with Focal (using Qt 5.12), I filed a PR draft: https://github.com/ros-planning/moveit_task_constructor/pull/392

Please review/test and report any remaining issues. I don't have a Humble installation on Focal to test myself. Thanks!

quarkytale, 2022-11-09:
Thank you so much for the quick fix @rhaschke, it worked for me!

rhaschke, 2022-11-10:
Closing the original issue as fixed. I cannot help with the launch issue you linked.

PR #392 (2022-11-10): Maintain compatibility to Qt 5.12 (for Focal)
Fixes #391?
closed 2022-11-10 by rhaschke