moveit/moveit_task_constructor #395: ClassLoader warning when removing one of multiple tasks

JSchaaf2, 2022-11-28:
I am not sure if I should ask this here or in the class loader but: I sometimes have multiple tasks, but when I remove one while still having another task I get this warning. I do not get this warning when I remove the last task.

```bash
[mtc-1] Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created
 by this loader exist in the heap! You should delete your objects before attempting to unload the library or 
destroying the ClassLoader. The library will NOT be unloaded.
[mtc-1]         at line 127 in ./src/class_loader.cpp
```

I am guessing the library is attempted to be unloaded when a task is removed, which fails because other tasks still use it. Because I don't get this warning if I remove the last task I guess I can safely ignore it. But I think it is not ideal to get such a scary looking warning if it is not actually a problem.

rhaschke, 2022-11-28:
Here is the right place to ask. Could you prepare a minimal example for this?
(I didn't observe that warning when using multiple tasks in the same process.)

JSchaaf2, 2022-12-01:
I thought I had a minimal example, it turns out I didn't. Apparently I still had a pointer to data of the task I was attempting to remove.

closed 2022-12-01 by JSchaaf2