moveit/moveit_task_constructor #396: CMake Warnings

captain-yoshi, 2022-11-28:
Updated MTC to latest commit https://github.com/ros-planning/moveit_task_constructor/commit/2cf4f8f0467c9482ca17c127a57575178d75ad2f and got unexpected warnings when building using `catkin build` on Ubuntu 20.04 and ROS Noetic. Also had to do a `git submodule update --init` to populate the pybind11 submodule. Also did a `catkin clean` on my workspace.

```bash
__________________________________________________________________________________________________________________________________________________________________________________________________________
Warnings   << moveit_task_constructor_core:symlink /home/captain-yoshi/ws/ros/mimik_ws/logs/moveit_task_constructor_core/build.symlink.002.log                                                            
Warning: Cannot symlink from /home/captain-yoshi/ws/ros/mimik_ws/devel/.private/moveit_task_constructor_core/lib/libmoveit_python_tools.so to existing file /home/captain-yoshi/ws/ros/mimik_ws/devel/lib/libmoveit_python_tools.so
Warning: Source hash: 90c7b0ce1b55b39fb365a3b4553c4ebb
Warning: Dest hash: aceb61c254b1b0a2445d1ab9f035fe5b
Warning: Cannot symlink from /home/captain-yoshi/ws/ros/mimik_ws/devel/.private/moveit_task_constructor_core/lib/python3/dist-packages/moveit/__init__.py to existing file /home/captain-yoshi/ws/ros/mimik_ws/devel/lib/python3/dist-packages/moveit/__init__.py
Warning: Source hash: 01bd41c1b260eac33aa0f3af16761b23
Warning: Dest hash: 09f1d6b001c3c83aa3e218a2bfa04bcf

__________________________________________________________________________________________________________________________________________________________________________________________________________
Warnings   << moveit_task_constructor_capabilities:cmake /home/captain-yoshi/ws/ros/mimik_ws/logs/moveit_task_constructor_capabilities/build.cmake.001.log                                                
CMake Warning at /home/captain-yoshi/ws/ros/mimik_ws/src/moveit_task_constructor/capabilities/CMakeLists.txt:34 (add_library):
  Cannot generate a safe runtime search path for target
  moveit_task_constructor_capabilities because there is a cycle in the
  constraint graph:

    dir 0 is [/home/captain-yoshi/ws/ros/mimik_ws/devel/.private/moveit_ros_move_group/lib]
    dir 1 is [/home/captain-yoshi/ws/ros/mimik_ws/devel/.private/moveit_ros_planning/lib]
    dir 2 is [/opt/ros/noetic/lib]
      dir 0 must precede it due to runtime library [libmoveit_move_group_capabilities_base.so.1.1.10]
      dir 1 must precede it due to runtime library [libmoveit_rdf_loader.so.1.1.10]
      dir 3 must precede it due to runtime library [libmoveit_ros_occupancy_map_monitor.so.1.1.10]
      dir 4 must precede it due to runtime library [libmoveit_python_tools.so]
      dir 5 must precede it due to runtime library [libmoveit_exceptions.so.1.1.10]
    dir 3 is [/home/captain-yoshi/ws/ros/mimik_ws/devel/.private/moveit_ros_occupancy_map_monitor/lib]
    dir 4 is [/home/captain-yoshi/ws/ros/mimik_ws/devel/.private/moveit_task_constructor_core/lib]
    dir 5 is [/home/captain-yoshi/ws/ros/mimik_ws/devel/.private/moveit_core/lib]
      dir 4 must precede it due to runtime library [libmoveit_python_tools.so]
    dir 6 is [/opt/ros/noetic/lib/x86_64-linux-gnu]
    dir 7 is [/home/captain-yoshi/ws/ros/mimik_ws/devel/.private/rviz_marker_tools/lib]

  Some of these libraries may not be found correctly.

cd /home/captain-yoshi/ws/ros/mimik_ws/build/moveit_task_constructor_capabilities; catkin build --get-env moveit_task_constructor_capabilities | catkin env -si  /usr/bin/cmake /home/captain-yoshi/ws/ros/mimik_ws/src/moveit_task_constructor/capabilities --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/captain-yoshi/ws/ros/mimik_ws/devel/.private/moveit_task_constructor_capabilities -DCMAKE_INSTALL_PREFIX=/home/captain-yoshi/ws/ros/mimik_ws/install . -DCMAKE_BUILD_TYPE=Release -DCMAKE_EXPORT_COMPILE_COMMANDS=1; cd -

```

rhaschke, 2022-11-28:
I had another idea for the origin of that problem: Maybe you have both MoveIt and MTC in your catkin workspace (we don't test this combo in our CI).

rhaschke, 2022-11-28:
Indeed, that seems to be the culprit. I was able to reproduce those warnings with such a workspace. I will try to fix that asap, but I'm very busy this week.

rhaschke, 2022-11-28:
The first symlink issue stems from the fact that both, MoveIt and MTC attempt to deploy a similar libmoveit_python_tools.so to the very same install location.

captain-yoshi, 2022-11-29:
Latest MoveIt commit https://github.com/ros-planning/moveit/commit/1f4435190f7498d746b8066fb72c956eebe8eeef.

> Maybe you have both MoveIt and MTC in your catkin workspace (we don't test this combo in our CI).

Yes I have both of them in my workspace.

rhaschke, 2023-02-28:
This was fixed with d95a2fc787dafa2dfa21bf49c77ffd0a68e7e485. The symlink warning will remain.

closed 2023-02-28 by rhaschke