moveit/moveit_task_constructor #399: MoveRelative does not respect cartesian lift up

captain-yoshi, 2022-11-29:
Stumbled upon a weird behavior where the cartesian solver with the `MoveRelative` stage seems to use a sampling based planner in one of the solutions (similar to the lift_object in the demo). Hard to replicate and may be due to my configuration of the task. 

```yaml
# solver parameters
type: cartesian
step_size: 0.01                    # step size between consecutive waypoints
jump_threshold: 1.5                # acceptable fraction of mean joint motion per step
min_fraction: 1.0                  # fraction of motion required for success
max_velocity_scaling_factor: 0.1
max_acceleration_scaling_factor: 0.1
```

https://user-images.githubusercontent.com/32679594/204626313-72ce6d74-2e10-4cb5-b5af-2bca8c43b15e.mp4

rhaschke, 2022-11-29:
For me, it looks like the IK solver switched from a right to left solution in the base and just performed linear interpolation between those.
When a sampling-based planner was used, this should be visible by quite different log output in the console.

captain-yoshi, 2022-11-29:
You are right, it makes more sense. Is this considered a bug ?

rhaschke, 2022-11-29:
Usually such "solutions" are rejected via `step_size` and `jump_threshold` parameters. Given your rather small values for those parameters, there might be a bug in MoveIt's `computeCartesianPath()`.
Which IK solver do you use? Can you reproduce the issue?

captain-yoshi, 2022-11-29:
I use `Track IK`. It is also hard to reproduce. I will try to reproduce and report back.

rhaschke, 2023-03-01:
Any news on this? Can we close this?

captain-yoshi, 2023-03-01:
Did not have the time to investigate more. Will try to reproduce with MoveIt and report back there.

closed 2023-03-01 by captain-yoshi