moveit/moveit_task_constructor #400: Release

vatanaksoytezer, 2022-11-29:
Is there anything blocking MTC to be released for noetic, rolling and humble? MTC seems to be used by many, and it would be great if we can give people access to binaries.

rhaschke, 2022-11-30:
@v4hn considered the API not yet stable and thus didn't agree to a release yet in the past.
Given, that this repo hasn't seen many commits in the last 2 years, this could be re-evaluated.
Before releasing though, we should address some existing and new issues (e.g. #396)

severin-lemaignan, 2025-10-15:
We are using it on humble, yes, and it works well, as far as I can tell.

rhaschke, 2025-10-15:
The smart_holder branch doesn't exist anymore, but the commit SHA the submodule is referencing does exist!

severin-lemaignan, 2025-10-15:
ahah! that's must be it, then.

closed 2025-10-16 by rhaschke