moveit/moveit_task_constructor #401: Segfault when creating multiple tasks

JSchaaf2, 2022-11-30:
I was creating a minimal example for another issue and finally stumbled on a something that looks a lot like another issue I had but never found a minimal example for. 

<details>
  <summary>modefied doTask() from pick and place tutorial</summary>

```c++
void MTCTaskNode::doTask() {
    task_ = createTask();
    task2 = createTask();

    try
    {
        task_.init();
    }
    catch (mtc::InitStageException& e)
    {
        RCLCPP_ERROR_STREAM(LOGGER, e);
        return;
    }
    if (!task_.plan(5))
    {
        RCLCPP_ERROR_STREAM(LOGGER, "Task planning failed");
        return;
    }

    try
    {
        task2.init();
    }
    catch (mtc::InitStageException& e)
    {
        RCLCPP_ERROR_STREAM(LOGGER, e);
        return;
    }
    if (!task2.plan(5))
    {
        RCLCPP_ERROR_STREAM(LOGGER, "Task planning failed");
        return;
    }

  return;
}
``` 
</details>

The above code causes a segfault in the collision checking of the IK wrapper for the second task and the following warning is very rarely given (I have only seen it once): [moveit_task_constructor_visualization.task_list_model]: unknown task: ...  . 
This might not be the only situation this happens but I have no other examples that are simple or fail reliably. The other cases also didn't show the warning so I don't know if that is just something I did wrong. 
Changing the code such that the `task2 = createTask();` line appears after the first one is initialised, or instead of being class members declaring the tasks like:
```c++
    mtc::Task task_ = createTask();
    mtc::Task task2 = createTask();
```
makes the segfault go away.

JafarAbdi, 2022-11-30:
Do you mind sharing the stacktrace when it segfaults?

> instead of being class members declaring the tasks like:
>
>    mtc::Task task_ = createTask();
>    mtc::Task task2 = createTask();
>
> makes the segfault go away.

I wonder if it's related to the class implementation, is it possible to share the whole class code?

JSchaaf2, 2022-11-30:
<details>
  <summary>This is backtrace I get</summary>

```bash
#0  0x00007ffff6382100 in ?? () from target:/lib/x86_64-linux-gnu/libfcl.so.0.7
#1  0x00007ffff63824a7 in ?? () from target:/lib/x86_64-linux-gnu/libfcl.so.0.7
#2  0x00007ffff6382545 in ?? () from target:/lib/x86_64-linux-gnu/libfcl.so.0.7
#3  0x00007ffff6382545 in ?? () from target:/lib/x86_64-linux-gnu/libfcl.so.0.7
#4  0x00007ffff637f32b in fcl::DynamicAABBTreeCollisionManager<double>::registerObjects(std::vector<fcl::CollisionObject<double>*, std::allocator<fcl::CollisionObject<double>*> > const&) () from target:/lib/x86_64-linux-gnu/libfcl.so.0.7
#5  0x00007ffff70ad676 in collision_detection::FCLObject::registerTo(fcl::BroadPhaseCollisionManager<double>*) () from target:/opt/ros/humble/lib/libmoveit_collision_detection_fcl.so.2.5.3
#6  0x00007ffff70b5f5f in collision_detection::CollisionEnvFCL::checkSelfCollisionHelper(collision_detection::CollisionRequest const&, collision_detection::CollisionResult&, moveit::core::RobotState const&, collision_detection::AllowedCollisionMatrix const*) const () from target:/opt/ros/humble/lib/libmoveit_collision_detection_fcl.so.2.5.3
#7  0x00007ffff7e40b50 in planning_scene::PlanningScene::checkCollision (acm=..., robot_state=..., res=..., req=..., this=0x5555567b4a10) at /opt/ros/humble/include/moveit/planning_scene/planning_scene.h:371
#8  moveit::task_constructor::stages::(anonymous namespace)::isTargetPoseCollidingInEEF (scene=..., robot_state=..., pose=..., link=<optimized out>, collision_result=0x7fffffff9640) at /home/jasper/MTC/src/moveit_task_constructor/core/src/stages/compute_ik.cpp:133
#9  0x00007ffff7e42553 in moveit::task_constructor::stages::ComputeIK::compute (this=0x555555e5ea30) at /home/jasper/MTC/src/moveit_task_constructor/core/src/stages/compute_ik.cpp:312
#10 0x00007ffff7ee9b1c in moveit::task_constructor::StagePrivate::runCompute (this=0x555555e5f740) at /home/jasper/MTC/src/moveit_task_constructor/core/include/moveit/task_constructor/stage_p.h:149
#11 0x00007ffff7ede00f in moveit::task_constructor::SerialContainer::compute (this=<optimized out>) at /home/jasper/MTC/src/moveit_task_constructor/core/src/container.cpp:684
#12 0x00007ffff7ee9b1c in moveit::task_constructor::StagePrivate::runCompute (this=0x555555e5d2a0) at /home/jasper/MTC/src/moveit_task_constructor/core/include/moveit/task_constructor/stage_p.h:149
#13 0x00007ffff7ede00f in moveit::task_constructor::SerialContainer::compute (this=<optimized out>) at /home/jasper/MTC/src/moveit_task_constructor/core/src/container.cpp:684
#14 0x00007ffff7ee9b1c in moveit::task_constructor::StagePrivate::runCompute (this=0x55555562cc30) at /home/jasper/MTC/src/moveit_task_constructor/core/include/moveit/task_constructor/stage_p.h:149
#15 0x00007ffff7f3a7b7 in moveit::task_constructor::Task::compute (this=0x55555562e600) at /home/jasper/MTC/src/moveit_task_constructor/core/include/moveit/task_constructor/container_p.h:190
#16 moveit::task_constructor::Task::plan (this=this@entry=0x55555562e600, max_solutions=max_solutions@entry=5) at /home/jasper/MTC/src/moveit_task_constructor/core/src/task.cpp:254
#17 0x0000555555561c1d in MTCTaskNode::doTask (this=0x55555562e5f0) at /home/jasper/MTC_minmal_example/src/mtc_tutorial.cpp:99
#18 0x000055555555d151 in main (argc=<optimized out>, argv=<optimized out>) at /usr/include/c++/11/bits/shared_ptr_base.h:1295
```
</details>

It is the tutorial code from the [pick and place example](https://moveit.picknik.ai/humble/doc/tutorials/pick_and_place_with_moveit_task_constructor/pick_and_place_with_moveit_task_constructor.html) in the moveit humble docs.

<details>
  <summary>My class looks like this</summary>

```c++
class MTCTaskNode
{
public:
  MTCTaskNode(const rclcpp::NodeOptions& options);

  rclcpp::node_interfaces::NodeBaseInterface::SharedPtr getNodeBaseInterface();

  void doTask();

  void setupPlanningScene();

private:
  // Compose an MTC task from a series of stages.
  mtc::Task createTask();
  mtc::Task task_;
  mtc::Task task2; //I added this line
  rclcpp::Node::SharedPtr node_;
};
```

</details>

<details>
  <summary>Here is the entire file to save having to look it up in the git or copy paste from the example</summary>

```c++
#include <rclcpp/rclcpp.hpp>
#include <moveit/planning_scene/planning_scene.h>
#include <moveit/planning_scene_interface/planning_scene_interface.h>
#include <moveit/task_constructor/task.h>
#include <moveit/task_constructor/solvers.h>
#include <moveit/task_constructor/stages.h>
#if __has_include(<tf2_geometry_msgs/tf2_geometry_msgs.hpp>)
#include <tf2_geometry_msgs/tf2_geometry_msgs.hpp>
#else
#include <tf2_geometry_msgs/tf2_geometry_msgs.h>
#endif
#if __has_include(<tf2_eigen/tf2_eigen.hpp>)
[... 323 lines cut ...]
int main(int argc, char** argv)
{
  rclcpp::init(argc, argv);

  rclcpp::NodeOptions options;
  options.automatically_declare_parameters_from_overrides(true);

  auto mtc_task_node = std::make_shared<MTCTaskNode>(options);
  rclcpp::executors::MultiThreadedExecutor executor;

  auto spin_thread = std::make_unique<std::thread>([&executor, &mtc_task_node]() {
    executor.add_node(mtc_task_node->getNodeBaseInterface());
    executor.spin();
    executor.remove_node(mtc_task_node->getNodeBaseInterface());
  });

  mtc_task_node->setupPlanningScene();
  mtc_task_node->doTask();

  spin_thread->join();
  rclcpp::shutdown();
  return 0;
}
```
</details>

JafarAbdi, 2022-11-30:
I thought we fixed this with https://github.com/ros-planning/moveit_task_constructor/commit/e923fbc0c6bd477d15532b335581734857d0eafc, I'll take a look later this week!

JSchaaf2, 2022-11-30:
> I thought we fixed this with [e923fbc](https://github.com/ros-planning/moveit_task_constructor/commit/e923fbc0c6bd477d15532b335581734857d0eafc), I'll take a look later this week!

Actually it look like the issue might be fixed, but I don't think it was e923fbc. I didn't properly clean everything and was still using 1e65027 (the commit after e923fbc), Using the latest commit in the ros2 branch that I can build without issues f28b6c4, the segfault from my example doesn't happen anymore.

rhaschke, 2022-11-30:
Between 1e65027 and f28b6c4 there was no actual code change. Hence, it is surprising to me that the former exhibits the segfault while the latter does not. You mention build issues with the latest ros2 branch. Could you please report them in a new issue?

JSchaaf2, 2022-12-06:
>  Could you please report them in a new issue?

I had some other issues with the moveit update, but I wanted to update before making the issue. I have updated to the latest ros2 and moveit version but I still have the same issues so I made #406.