moveit/moveit_task_constructor #405: Error when displaying the last waypoint of a MoveRelative stage

captain-yoshi, 2022-12-04:
This seems to happen only in RViz and only on the `lower object` sub-task , executed on a real robot without problem. Similar to the demo except that there's a `ModifyPlanningScene` between the MoveRelative and the GeneratePlacePose stages. The lift up does not exhibit this behavior.

```cpp
    /****************************************************
  ---- *               Allow Collision (hand object)   *
     ***************************************************/
    {
      auto stage = std::make_unique<stages::ModifyPlanningScene>("allow collision (hand,object)");
      stage->allowCollisions("thermocycler", attached_object_names_, false);
      place->insert(std::move(stage));
    }
```

https://user-images.githubusercontent.com/32679594/205476330-0bb1ea46-89cc-40ca-a4d1-e08b3649e0cf.mp4

rhaschke, 2022-12-04:
I am not sure what exactly is the error here. Could you please elaborate a little more?
Note that the `ModifyPlanningScene` state is applied in reverse order automatically. Thus, if you want to allow collisions, you should pass `true`, not `false`.

captain-yoshi, 2022-12-04:
> Could you please elaborate a little more?

The last waypoint seems to jump (waypoint 27 -> 28) back to the start waypoint when visualizing with the slider.

> Thus, if you want to allow collisions, you should pass true, not false.

Following https://github.com/ros-planning/moveit_task_constructor/issues/349#issuecomment-1085002152, I think I got to set this to `False` to actually allow the collision. If I pass `True` the lower object fails because it detects the collision.

rhaschke, 2022-12-04:
I see. Could you try to provide a _minimal_ example, please?

captain-yoshi, 2022-12-04:
@rhaschke Invited you to a minimal example with configuration/running step.

rhaschke, 2023-02-28:
Sorry for the delay on this. I was too busy with other stuff and just came across this again when cleaning up my inbox.
See #432 for a potential fix. Could you please review/test?

captain-yoshi, 2024-02-08:
I have another problem maybe related to this issue. Will see if #432 fixes that now that some fixes has been made with MoveIt.

Hypothesis: Because of the last erroneous waypoint, the connect stage is able to connect to wrong robot states.
- 0 to 4 sec: the RViz display bug where the end waypoint has wrong state (`move to pick` stage).
- 10 tio 20 sec: Connected wrongly to the approach object stage. Wrong robot state. Note that the wrong state is from the `approach object` stage.

https://github.com/ros-planning/moveit_task_constructor/assets/32679594/f61264e8-b958-4cd7-b996-87a2c9be5261

captain-yoshi, 2024-02-08:
Still got both bugs with #432. Using MoveIt latest commit [ros-planning/moveit/8f39ef7](https://github.com/ros-planning/moveit/commit/8f39ef78bcae5940f5aad097c3c2fd4a4667b224).

rhaschke, 2024-02-09:
I didn't yet try this PR since https://github.com/ros-planning/moveit/pull/3538 was merged. But @v4hn claimed that the issues should have been resolved. Regarding your first question: The bug stated here does not effect the planning itself. This is "just" a visualization bug.

commit 795bde7b (2024-03-09): Fix Solution::fillMessage() (#432)

The scene_diff field usually describes PlanningScene changes of the end scene relative to the start scene.
For backwards planning, this direction is reversed: the start scene is derived from the end scene.
Thus, we need to generate a full planning scene message for the end scene if planning progressed backwards.

Fixes #405.
PR #432 (2024-03-09): Fix Solution::fillMessage()
Need to generate full planning scene message if planning progressed backwards: The scene_diff field describes PlanningScene changes of the end scene relative to the start scene, while during planning it might be inversed.
Seemingly fixes #405. I would like to add some unit tests before merging.
I also noticed that `ModifyPlanningScene::(add|remove)Object` doesn't yet work in backwards mode. Shall we support this or just issue an error? To reverse, `removeObject()` would require the full `CollisionObject` description (to actually create the object), while currently ( [... cut]
closed 2024-03-09 by rhaschke