moveit/moveit_task_constructor #408: MTC segfaults when trying to access failures from stages nested inside containers

Schteve-Earl, 2022-12-06:
In the below example, I am attempting to move the panda arm with a motion that is kinematically impossible, and then extract the comment from the error. I am able to access the failures when a stage is added directly to a task, however, the program is segfaulting when I attempt to access the failures of a stage that has been inserted into a serial container:
```
#include <moveit/task_constructor/task.h>

#include <moveit/task_constructor/stages/fixed_state.h>
#include <moveit/task_constructor/solvers/cartesian_path.h>
#include <moveit/task_constructor/solvers/joint_interpolation.h>
#include <moveit/task_constructor/stages/move_to.h>
#include <moveit/task_constructor/stages/move_relative.h>
#include <moveit/task_constructor/stages/connect.h>

#include <ros/ros.h>
#include <moveit/planning_scene/planning_scene.h>

using namespace moveit::task_constructor;

Task createTask() {
	Task t;
	t.stages()->setName("Cartesian Path");

	const std::string group = "panda_arm";
	const std::string eef = "hand";

	// create Cartesian interpolation "planner" to be used in various stages
	auto cartesian_interpolation = std::make_shared<solvers::CartesianPath>();
	// create a joint-space interpolation "planner" to be used in various stages
	auto joint_interpolation = std::make_shared<solvers::JointInterpolationPlanner>();

	// start from a fixed robot state
	t.loadRobotModel();
	auto scene = std::make_shared<planning_scene::PlanningScene>(t.getRobotModel());
	{
		auto& state = scene->getCurrentStateNonConst();
		state.setToDefaultValues(state.getJointModelGroup(group), "ready");

		auto fixed = std::make_unique<stages::FixedState>("initial state");
		fixed->setState(scene);
		t.add(std::move(fixed));
	}

	{
		auto stage = std::make_unique<stages::MoveRelative>("y -0.8", cartesian_interpolation);
		stage->setGroup(group);
		geometry_msgs::Vector3Stamped direction;
		direction.header.frame_id = "world";
		direction.vector.y = -0.8;
		stage->setDirection(direction);
		t.add(std::move(stage));
	}

	{  // move from reached state back to the original state, using joint interpolation
		// specifying two groups (arm and hand) will try to merge both trajectories
		stages::Connect::GroupPlannerVector planners = { { group, joint_interpolation }, { eef, joint_interpolation } };
		auto connect = std::make_unique<stages::Connect>("connect", planners);
		t.add(std::move(connect));
	}

	{  // final state is original state again
		auto fixed = std::make_unique<stages::FixedState>("final state");
		fixed->setState(scene);
		t.add(std::move(fixed));
	}

	return t;
}

Task createTaskWithContainer() {
	Task t;
	t.stages()->setName("Cartesian Path");

	const std::string group = "panda_arm";
	const std::string eef = "hand";

	// create Cartesian interpolation "planner" to be used in various stages
	auto cartesian_interpolation = std::make_shared<solvers::CartesianPath>();
	// create a joint-space interpolation "planner" to be used in various stages
	auto joint_interpolation = std::make_shared<solvers::JointInterpolationPlanner>();

	// start from a fixed robot state
	t.loadRobotModel();
	auto scene = std::make_shared<planning_scene::PlanningScene>(t.getRobotModel());
	{
		auto& state = scene->getCurrentStateNonConst();
		state.setToDefaultValues(state.getJointModelGroup(group), "ready");

		auto fixed = std::make_unique<stages::FixedState>("initial state");
		fixed->setState(scene);
		t.add(std::move(fixed));
	}

	{
		auto container = std::make_unique<SerialContainer>("container");
		container->storeFailures();
		{
			auto stage = std::make_unique<stages::MoveRelative>("y -0.8", cartesian_interpolation);
			stage->storeFailures();
			stage->setGroup(group);
			geometry_msgs::Vector3Stamped direction;
			direction.header.frame_id = "world";
			direction.vector.y = -0.8;
			stage->setDirection(direction);
			container->insert(std::move(stage));
		}
		t.add(std::move(container));
	}

	{  // move from reached state back to the original state, using joint interpolation
		// specifying two groups (arm and hand) will try to merge both trajectories
		stages::Connect::GroupPlannerVector planners = { { group, joint_interpolation }, { eef, joint_interpolation } };
		auto connect = std::make_unique<stages::Connect>("connect", planners);
		t.add(std::move(connect));
	}

	{  // final state is original state again
		auto fixed = std::make_unique<stages::FixedState>("final state");
		fixed->setState(scene);
		t.add(std::move(fixed));
	}

	return t;
}

int main(int argc, char** argv) {
	ros::init(argc, argv, "mtc_tutorial");
	// run an asynchronous spinner to communicate with the move_group node and rviz
	ros::AsyncSpinner spinner(1);
	spinner.start();

	// This works, stage "y -0.8" fails as expected and the comment is printed as expected
	auto task = createTask();
	if (task.plan())
		task.introspection().publishSolution(*task.solutions().front());
	else {
		ROS_ERROR_STREAM("y -0.8 failure is " << task.stages()->findChild("y -0.8")->failures().front()->comment());
	}

	// This segfaults because it is unable to find the child associated that is locked behind the serial container
	task = createTaskWithContainer();
	if (task.plan())
		task.introspection().publishSolution(*task.solutions().front());
	else {
		ROS_ERROR_STREAM("y -0.8 failure is " << task.stages()->findChild("y -0.8")->failures().front()->comment());
	}

	ros::waitForShutdown();  // keep alive for interactive inspection in rviz
	return 0;
}
```
Running the program with the demo.launch file in /demo results in the following terminal output:
[image]

Is this a bug, or is intended for errors to be inaccessible when placed inside a container? If the latter, is there any way to extract the failure information from a stage that has been inserted inside a container?

JafarAbdi, 2022-12-06:
You need to check the size of the `failures()` before accessing it, as you can see in the image it didn't generate any solutions, `connect` is the one that failed

Schteve-Earl, 2022-12-06:
@JafarAbdi The code is faulting when I attempt to access the result of the findChild function in the following code, not just because I was accessing an empty vector
```
	// This segfaults because it is unable to find the child associated that is locked behind the serial container
	task = createTaskWithContainer();
	if (task.plan())
		task.introspection().publishSolution(*task.solutions().front());
	else {
	// Segfaults when trying to access the result of the findChild() function
		ROS_ERROR_STREAM("Size of y -0.8 failures is " << task.stages()->findChild("y -0.8")->numFailures());
		ROS_ERROR_STREAM("y -0.8 failure is " << task.stages()->findChild("y -0.8")->failures().front()->comment());
	}
```
And here is rviz before the crash:
[image]

As you can see, the "y -0.8" stage is failing, but I can't access that information.

rhaschke, 2022-12-06:
You can't access a child state within a container directly. You need to specify the full path to this child:
`findChild("container/y -0.8")`

Obviously, rviz cannot access anything if the main MTC program crashed...

Schteve-Earl, 2022-12-06:
@rhaschke You're correct, I was just providing the rviz output to show that connect wasn't the failing stage.

Otherwise, this solved my issue. Thanks for your help! Sidenote, is there a location where this is documented? If not, I'll gladly create an error handling demo for this repo if there is any interest so others don't run into the same issue.

closed 2022-12-06 by Schteve-Earl

rhaschke, 2022-12-06:
Documentation is still limited. You are welcome to contribute though.
Have a look at https://ros-planning.github.io/moveit_task_constructor. Docs are generated essentially from code (via doxygen and sphinx).