moveit/moveit_task_constructor #421: [ROS2] Retrieve solution and execute with MoveItCpp

g-argyropoulos, 2023-01-30:
Hello,

Is it possible to execute a task solution with MoveItCpp ?
I am trying to end up with a _robot_trajectory::RobotTrajectoryPtr_  that can be used as input to the execute member function of moveitcpp. 

Thank you

rhaschke, 2023-01-30:
MTC's solutions are richer than those accepted by the default MoveIt execution framework, particularly planning scene changes are performed in addition to the actual robot trajectories. As the MTC code was written before MoveItCpp, direct support is not yet available. However, a PR implementing this is highly welcome. I suggest a call like `Task::execute(MoveItCpp& m, const Solution&)`. Have a look into [`ExecuteTaskSolutionCapability::goalCallback`](https://github.com/ubi-agni/moveit_task_constructor/blob/10fe2729380eb080b61125c1df46334cf4c18b3f/capabilities/src/execute_task_solution_capability.cpp#L93) for the necessary code.

rhaschke, 2023-05-04:
I implemented this in #459.