moveit/moveit_task_constructor #422: Pick-place planning continues after all solutions of Approach stage failed

Smail8, 2023-02-05:
Hello,

I noticed that sometimes in a Pick-Place planning task, when none of the solutions to the Approach stage are successful, MTC continues to develop the solutions of the Place container, even if the whole task can already be marked as failure.

Here is a video showing the issue. Planning starts normally, but at 00:19, after all Approach solutions have been exhausted with none being successful, the planning continues :

https://user-images.githubusercontent.com/43912811/216804098-4f07fd7b-0654-4ead-bf71-643d37b705d9.mov

My application needs fast planning times, so I can't afford the overhead caused by this bug. I suspect that the problem comes from the fact that GeneratePlacePose monitors the AttachObject stage, but I can't find a solution or a work around to avoid this. Do you have any idea how to solve this ?

Note: I modified the library code a bit to fit my needs, but I don't think I modified anything that might cause this issue.

v4hn, 2023-02-05:
The concrete issue you report is a known shortcoming of the current pruning approach: #313 .
A potential workaround might monitor the whole pick container instead of the `AttachObject` stage and utilize the end scene of its solutions.
This will generate different search trees inside the system, but will eventually generate the same set of valid solutions.

Smail8, 2023-02-06:
Thank you for your response. The suggested workaround worked.

closed 2023-02-06 by Smail8