moveit/moveit_task_constructor #427: Release issue: cannot install moveit/__ini__.py from different namespace packages

rhaschke, 2023-02-16:
I fixed most release issues for ROS1, but there is one remaining, again related to [`python` namespace packages](https://packaging.python.org/guides/packaging-namespace-packages/#pkgutil-style-namespace-packages): Debians from different packages attempt to install their (identical) `moveit/__init__.py` to the very same target location in /opt/ros, which [fails](https://build.ros.org/job/Nbin_db_dB64__moveit_task_constructor_capabilities__debian_buster_amd64__binary/6/console) with:
```
03:52:20   3364 'Unpacking ros-noetic-moveit-task-constructor-core (0.1.1-1buster.20230216.023555) ...'
03:52:20   3365 'dpkg: error processing archive /tmp/apt-dpkg-install-kkWIOZ/709-ros-noetic-moveit-task-constructor-core_0.1.1-1buster.20230216.023555_amd64.deb (--unpack):'
03:52:20   3366 ' trying to overwrite '/opt/ros/noetic/lib/python3/dist-packages/moveit/__init__.py', which is also in package ros-noetic-moveit-core 1.1.11-1buster.20230216.000323'
```
I'm not sure how this issue is handled in .debs in general, but I see two options only:
- Suppress deploying this file in downstream packages like MTC and just rely on moveit_core's file to be present
- Explicitly allow overwriting of this file (as we know that they _should_ all be identical)

I prefer the latter method. 
@jspricke, do you have a hint on how to realize that with the debian release process or `bloom` actually? 
@JafarAbdi, you [offered your help](https://github.com/ros-planning/moveit_task_constructor/pull/425#issuecomment-1426750845) in the release process. As I don't have time to handle this issue in the upcoming days, I would be grateful for your contribution.

rhaschke, 2023-02-16:
@peterdavidfagan, you will run into this issue with https://github.com/ros-planning/moveit2/pull/1546 as well...

jspricke, 2023-02-16:
> 03:52:20   3364 'Unpacking ros-noetic-moveit-task-constructor-core (0.1.1-1buster.20230216.023555) ...'
> 03:52:20   3365 'dpkg: error processing archive [...]

You should not install the file as it is guaranteed to be there by the 
package dependency. Something like if(CMAKE_INSTALL_PREFIX STREQUAL 
"/opt") in your CMakeLists.txt could do.

rhaschke, 2023-02-16:
> Something like if(CMAKE_INSTALL_PREFIX STREQUAL "/opt") in your CMakeLists.txt could do.

I was hoping for a "cleaner" solution explicitly stating this exception in the debian release files or removing the file before packaging the .deb, e.g. with a bloom patches directory.

jspricke, 2023-02-16:
>> Something like if(CMAKE_INSTALL_PREFIX STREQUAL "/opt") in your CMakeLists.txt could do.
>
>I was hoping for a "cleaner" solution explicitly stating this exception in the debian release files or removing the file before packaging the .deb, e.g. with a bloom patches directory.

I consider the CMakeLists.txt option cleaner then adding bloom patches. 
Patches are kind of a fork of the project and need extra maintenance. I 
assume that the file is just there for convenience but not actually part 
of the project, as it belongs to moveit_core. So it should be the 
exception to install it.

rhaschke, 2023-02-16:
Ok, we would need then an exception for `setup.py` (instead of `cmake`) as the python stuff is installed via the former.

peterdavidfagan, 2023-02-16:
> @peterdavidfagan, you will run into this issue with https://github.com/ros-planning/moveit2/pull/1546 as well...

Thanks for the heads up on this @rhaschke, I will take a look into this once I get the chance to.

v4hn, 2023-02-16:
It seems awfully bad practice to install different projects into the same package name altogether.
What happens if moveit and MTC are build in different (overlaying) workspaces?
I believe the `moveit` package/folder generated by mtc will overlay the one in the parent workspace.
Did anyone validate this use case actually works?

rhaschke, 2023-02-16:
> What happens if moveit and MTC are build in different (overlaying) workspaces?

Surely, I tested this. That's the idea of [python's namespaced packages](https://packaging.python.org/guides/packaging-namespace-packages/#pkgutil-style-namespace-packages). It works, with [this](https://github.com/ros-planning/moveit/pull/2910#issuecomment-1431600602) unrelated exception. 
We cannot install into a different location as this would result in a python package named `location.moveit.task_constructor`.

JafarAbdi, 2023-02-18:
I looked into this today, and I couldn't find a way to do it in setup.py. I think bloom patches is the only option

rhaschke, 2023-02-23:
Declaring an `exclude` in `MANIFEST.in` actually allows excluding files from installing:
https://stackoverflow.com/questions/26911045/is-it-possible-to-exclude-certain-files-when-building-a-wheel-with-setup-py

However, we cannot do that conditionally only for the debian build.

rhaschke, 2023-02-23:
Fix: 6671aedeb722f20ddf18a3b95394d1086b51d951
https://github.com/ros/rosdistro/pull/36284

rhaschke, 2023-03-06:
Finally, the [release builds succeeded](http://repositories.ros.org/status_page/ros_noetic_default.html?q=moveit_task_constructor) :rocket: 
I will try to handle the ROS2 release over next weekend. @JafarAbdi, are there any open issues for ROS2?

closed 2023-03-06 by rhaschke