moveit/moveit_task_constructor #433: Task execution failed and returned: -4

mrceki, 2023-03-02:
Hi!
I'm working on a robot with Qbhand Softhand gripper and trying to perform a pick and place. But when it comes to gripper action, i get this error and than mtc stops:

- Task execution failed and returned: -4 

I think thats because of grippers hardware interface. Can you suggest any solution?

Thanks.

mrceki, 2023-03-02:
I'm using FollowJointTrajectory.

mrceki, 2023-03-06:
I'm trying debug the bug and found something, but not sure is that the problem. When i listen `/qbhand/control/qbhand_synergy_trajectory_controller/follow_joint_trajectory/goal` topic, like 40-50 points are generated by move group. Buy when i run MTC, just 2-3 points are generated. Do you think that show us the problem?
```
header: 
  seq: 2
  stamp: 
    secs: 1678090679
    nsecs: 977593438
  frame_id: ''
goal_id: 
  stamp: 
    secs: 1678090679
    nsecs: 977593926
  id: "/move_group-3-1678090679.977593926"
goal: 
  trajectory: 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: "base_link"
    joint_names: 
      - qbhand_synergy_joint
    points: 
      - 
        positions: [0.00905263157894737]
        velocities: [0.0]
        accelerations: [0.0]
        effort: []
        time_from_start: 
          secs: 0
          nsecs:         0
      - 
        positions: [9.14573384448886e-06]
        velocities: [0.0]
        accelerations: [0.9454074126043495]
        effort: []
        time_from_start: 
          secs: 0
          nsecs: 138316321
  path_tolerance: []
  goal_tolerance: []
  goal_time_tolerance: 
    secs: 0
    nsecs:         0
---
```

rhaschke, 2023-03-06:
Did you check the generated trajectory in MTC's rviz GUI? Are there two waypoints only as well?

mrceki, 2023-03-06:
No, there's 51 waypoints.
Another tip is, joint states of gripper published to `/qbhand/control/joint_states`. Will this cause a problem?

rhaschke, 2023-03-06:
> joint states of gripper published to `/qbhand/control/joint_states`. Will this cause a problem?

All of ROS including MoveIt expects them on `/joint_states`. But if it works from move_group / MotionPlanning Display, that's _probably_ not the culprit.

mrceki, 2023-03-06:
Most likely this is the problem. Because this way I can't use gripper properly. The gripper moves with a value between 0-1 and let's assume a scenario like this:

First I move the joint to `0.5`, and then I want to move it to `0.8`, but instead of going from `0.5` to `0.8`, it opens all the way and moves to `0.8.` 
[image]

This problem is solved if I remap the topic `/qbhand/control/joint_states to /joint_states` and it can move from `0.5` to `0.8.`
[image]

I guess the problem will be solved if these two topics are merged, but don't know how to merge these two topics for full solution because `/joint_states` is published by hardware interface.

mrceki, 2023-03-06:
> You could consider using the joint_state_publisher with a suitably configured source_list parameter ([docs](http://wiki.ros.org/joint_state_publisher#Parameters)) to coalesce JointState messages coming in over multiple topics and republish them to /joint_states.

Since this is a research project, it does not need to be consumer-oriented.

I remapped /joint_states to /robot/joint_states and created new joint_state_publisher:

```
    <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher">
      <param name="rate" value="100"/>
      <rosparam param="source_list">[/qbhand/control/joint_states]</rosparam>
      <rosparam param="source_list">[/robot/joint_states]</rosparam>
    </node>
```
After doing this, the topic message I wrote on the 2nd line is received as blank.
```
header: 
  seq: 3107
  stamp: 
    secs: 1678102158
    nsecs: 388141393
  frame_id: ''
name: 
  - s1j
  - s2j
  - s3j
  - ej
  - rollj
  - pitchj
  - yawj
  - qbhand_synergy_joint
  - qbhand_thumb_knuckle_joint
  - qbhand_thumb_proximal_joint
  - qbhand_thumb_proximal_virtual_joint
  - qbhand_thumb_distal_joint
  - qbhand_thumb_distal_virtual_joint
  - qbhand_index_knuckle_joint
  - qbhand_index_proximal_joint
  - qbhand_index_proximal_virtual_joint
  - qbhand_index_middle_joint
  - qbhand_index_middle_virtual_joint
  - qbhand_index_distal_joint
  - qbhand_index_distal_virtual_joint
  - qbhand_middle_knuckle_joint
  - qbhand_middle_proximal_joint
  - qbhand_middle_proximal_virtual_joint
  - qbhand_middle_middle_joint
  - qbhand_middle_middle_virtual_joint
  - qbhand_middle_distal_joint
  - qbhand_middle_distal_virtual_joint
  - qbhand_ring_knuckle_joint
  - qbhand_ring_proximal_joint
  - qbhand_ring_proximal_virtual_joint
  - qbhand_ring_middle_joint
  - qbhand_ring_middle_virtual_joint
  - qbhand_ring_distal_joint
  - qbhand_ring_distal_virtual_joint
  - qbhand_little_knuckle_joint
  - qbhand_little_proximal_joint
  - qbhand_little_proximal_virtual_joint
  - qbhand_little_middle_joint
  - qbhand_little_middle_virtual_joint
  - qbhand_little_distal_joint
  - qbhand_little_distal_virtual_joint
position: [0.43009218847200054, 0.03647942013875949, 0.017900567894321415, -0.48762466102499347, -0.07788767452587626, -0.03474690654168362, 0.02379476134626788, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
velocity: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
effort: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
```

gavanderhoorn, 2023-03-06:
>     <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher">
>       <param name="rate" value="100"/>
>       <rosparam param="source_list">[/qbhand/control/joint_states]</rosparam>
>       <rosparam param="source_list">[/robot/joint_states]</rosparam>
>     </node>

this sets `source_list` twice. Once with the topic `/qbhand/control/joint_states` and once with `/robot/joint_states`. Last setting wins in `roslaunch` XML, so this does not do what you want to do I believe.

`source_list` should be a list, each element being a topic name.

mrceki, 2023-03-06:
Thanks for your help @gavanderhoorn ! For having the same problem, syntax should be like this:
```
    <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher">
      <param name="rate" value="100"/>
      <rosparam param="source_list">
        - /qbhand/control/joint_states
        - /joint_states
      </rosparam>    
    </node>
```
But this unfortunately did not solve my problem. When I start the program while the gripper is closed, the gripper opens, but when it goes to grasp pose, gripper won't close.

gavanderhoorn, 2023-03-06:
I don't understand what you mean, but IIUC, adding `/joint_states` to `source_list` is probably not what you want to do.

You really should only list topics which should be considered *input* topics. `/joint_states` is an *output* topic for JSP.

mrceki, 2023-03-06:
I see your point and updated my comment.

mrceki, 2023-03-06:
If you can think of anything about solving the problem, I would be glad.

mrceki, 2023-03-06:
Sorry to bother, but there's another warning:

```
[ WARN] [1678110646.109372700]: Dropping all 1 trajectory point(s), as they occur before the current time.
Last point is 0.000000s in the past.
[ WARN] [1678110646.109536920]: Controller 'arm/arm_position_controller' failed with error INVALID_GOAL: Dropping all 1 trajectory point(s), as they occur before the current time.
Last point is 0.000000s in the past.
[ WARN] [1678110646.109606746]: Controller handle arm/arm_position_controller reports status FAILED
```

rhaschke, 2023-03-06:
I have seen these messages from the ROS trajectory controller, when the trajectory is essentially "empty", i.e. the target waypoint matches the current robot state (i.e. no motion needed). The generated trajectory in that case has a goal time of zero, which seems to confuse the trajectory controller. It's on my TODO list to workaround that issue on the MTC/MoveIt side.

mrceki, 2023-03-07:
Do you have any idea to save the day?

rhaschke, 2023-03-07:
I reshuffled my task queue for you and provided a workaround: https://github.com/ros-planning/moveit_task_constructor/pull/438
If that saved your day, please consider [donating](https://github.com/sponsors/rhaschke).