moveit/moveit_task_constructor #441: MTC Stages

juandavid212, 2023-03-14:
Hello, 
Is there a way to control the amount of time the robot stays in a particular stage?
-I want to generate a trajectory where the robot passes thru specific stages without stopping. 

-How can I implement a stage that pauses the execution while waiting for a flag to continue the execution?

rhaschke, 2023-03-14:
> Is there a way to control the amount of time the robot stays in a particular stage?

Right now: no. 
But, I'm currently reworking the whole execution code to allow for exactly such things.  
Any [donation](https://github.com/sponsors/rhaschke) to sponsor that work is highly welcome.

matthias88, 2023-12-19:
Hi everyone, are there any news about this topic? Due to the target cycle time i would like to exeute the whole task without "stops" between the stages. Are there any workarounds? Possibly using a moveit planning request adapter to fix start state of each stage or convert path into time parametrized trajectory without "stops"?

rhaschke, 2023-12-19:
If you just want to realize a smooth execution of a complex task, you could time-reparameterize the combined trajectories of your favorite solution.